pimoroni/pimoroni-pico #749: VL53L5CX MicroPython Possible Memory Leak

jjpendlebury, 2023-04-12:
Hi All,

While running the Pi Pico MicroPython VL53L5CX demo found [here](https://github.com/pimoroni/pimoroni-pico/blob/7b7352130f1980d523c0f446656a40eaeaee9833/micropython/examples/breakout_vl53l5cx/vl53l5cx_demo.py), the Pi Pico freezes after a certain amount of reads. 

Using `micropython.mem_info()` I can see the amount of free memory steadily decreasing as the demo continues running. This on the latest Pimoroni Pi Pico firmware - V1.19.18 (as of the time of writing).

https://github.com/pimoroni/pimoroni-pico/blob/7b7352130f1980d523c0f446656a40eaeaee9833/micropython/examples/breakout_vl53l5cx/vl53l5cx_demo.py#L26-L36

Adding a call to `gc.collect()` at the end of the loop solves the problem, letting the demo continue running.

I'm new to MicroPython/RPi, so this could just a be a known thing, in which case, my apologies.

ZodiusInfuser, 2023-04-13:
Hi @jjpendlebury,

Micropython is lazy in its cleaning up of memory, so I suspect what you are seeing from `mem_info()` is just old versions of variables (such as that `data` one) being kept around. Even the act of querying the amount of memory and printing it will cause your memory to go down. Eventually though the GC should kick in when memory gets to a low enough point that it needs to allocate something larger than the remaining space. If that fails then you usually receive an out of memory error.

When you are running the VL53L5CX demo and your Pi Pico freezes you do get such an error? Are you able to recover the Pi Pico by terminating the program or do you have to unplug the power?

What version of our Micropython build are you using? If it's the latest then it may be worth trying an older version, as there were some big memory changes in the last few, and perhaps an issue with the VL53L5CX was introduced.

jjpendlebury, 2023-04-13:
@ZodiusInfuser Thanks for the reply! I did see that about MicroPython, hence me being unsure if this was an actual issue or not. It's also somewhat frustrating that MicroPython seems to lack some of the tools for memory profiling like I'm used to (Or I just haven't found them yet!).

I'm running the latest release from [here](https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.18), v1.19.18. I didn't see any out of memory error within Thonny's Terminal when the lock-up happened, in-fact there wasn't any error message at all, the board goes completely unresponsive.

Terminating the program wouldn't work, I'd have to de-power the board, plug it back in, and then I could halt execution through Thonny. While working with this demo I'd had to do that every time to program the board. 

Since re-flashing the firmware, the issue seems to have gone away, and automatic garbage collection is working again. Whatever the issue was, it seems like it was on my end. Something about my specific software/hardware combination preventing automatic garbage collection from working. I'm happy to close this one, thank you for you time and your help!

closed 2023-04-13 by jjpendlebury