pimoroni/pimoroni-pico #752: Inaccurate version info in Pimoroni specific micropython images

cpottle9, 2023-04-14:
When I run thonny with a Pimoroni micropython uf2 file I see: 
```
MicroPython 294098d-dirty on 2023-03-08
```

I wrote a little bash script named mp_uf2_version.sh to extract version info directly from the uf2 file.
```
#!/bin/bash
for uf2 in "$@"
do
    version=`strings "$uf2" | grep 'MicroPython.*202[0-9]'`
    printf "%-60s: %s\n" "$uf2" "$version"
done
```

When I run this against the Pimoroni uf2 files in my Downloads directory I see:
```
~/Downloads $ mp_uf2_version.sh pimoroni-*uf2
pimoroni-badger2040w-v0.0.2-micropython.uf2                 : 3.4.0; MicroPython 38e7b84-dirty on 2023-03-30
pimoroni-badger2040w-v1.19.16-micropython.uf2               : 3.4.0; MicroPython 294098d-dirty on 2023-03-08
pimoroni-picow_inky_frame-v1.19.16-micropython.uf2          : 3.4.0; MicroPython 294098d-dirty on 2023-03-08
pimoroni-picow_inky_frame-v1.19.18-micropython.uf2          : 3.4.0; MicroPython 38e7b84-dirty on 2023-03-27
```
The word dirty indicates an uncontrolled version. 
The git commit hash is from the micropython repo. 
Not from any repo owned by Pimoroni.

But, for the 'vanilla' uf2 files for pico w board I see:
```
~/Downloads $ mp_uf2_version.sh rp2-pico-w-20230*uf2
rp2-pico-w-20230201-unstable-v1.19.1-850-gfe2a8332f.uf2     : 3.4.0; MicroPython v1.19.1-850-gfe2a8332f on 2023-02-01
rp2-pico-w-20230308-unstable-v1.19.1-939-ga234aa44f.uf2     : 3.4.0; MicroPython v1.19.1-939-ga234aa44f on 2023-03-08
```
The word dirty is missing and it contains the git tag and commit hash.

It would be good if your uf2 files contained the git tags and the git hash.
That way users and your folks could easily tell which version is running.

I took a look at makeversionhdr.py. 
MICROPY_GIT_TAG and MICROPY_GIT_HASH can be overridden using environment variables.
The important information is in MICROPY_GIT_TAG.
In the current builds the value of MICROPY_GIT_HASH  is not used.
You can set environment variable MICROPY_GIT_TAG with version info from the badger2040 or pimoroni-pico repo as appropriate.

Gadgetoid, 2023-04-26:
Well this is going well for me :laughing: 

```
bash version.sh *.uf2
pimoroni-cosmic_unicorn-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 3.4.0; MicroPython $MICROPYTHON_VERSION, cosmic_unicorn c1db2da33a884e57a4aa175c8479a8fca1c230ca on 2023-04-26
pimoroni-enviro-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 3.4.0; MicroPython $MICROPYTHON_VERSION, enviro c1db2da33a884e57a4aa175c8479a8fca1c230ca on 2023-04-26
pimoroni-galactic_unicorn-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 
pimoroni-inky_frame-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 3.4.0; MicroPython $MICROPYTHON_VERSION, inky_frame c1db2da33a884e57a4aa175c8479a8fca1c230ca on 2023-04-26
pimoroni-pico-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 
pimoroni-picolipo_16mb-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 
pimoroni-picolipo_4mb-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 
pimoroni-picow-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 3.4.0; MicroPython $MICROPYTHON_VERSION, picow c1db2da33a884e57a4aa175c8479a8fca1c230ca on 2023-04-26
pimoroni-tiny2040-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 
pimoroni-tufty2040-c1db2da33a884e57a4aa175c8479a8fca1c230ca-micropython.uf2: 
```

Gadgetoid, 2023-05-25:
This is now rolled out and seems to work.

closed 2023-05-25 by Gadgetoid