pimoroni/pimoroni-pico #765: FR: Add a monospace bitmap font to PicoGraphics

LionsPhil, 2023-04-29:
Looking at their definitions in https://github.com/pimoroni/pimoroni-pico/tree/cb5db1d5944d0b6adc9ef9b7c014749b03301deb/libraries/bitmap_fonts , all the available fonts for PicoGraphics---bitmap6, 8, and 14_outline---are proportionally spaced (variable width).

I'd like a monospace font, so that debug readouts and the like in Tufty2040 scripts that have rapidly-updating numbers don't jump about (since "1" is narrower than other digits). A monospace variant of bitmap8 would be just fine, for example.

Being able to provide the font data via the Python API would natch solve this, and "can I have another font please" in the general case. It looks...plausible? to implement that, with perhaps some awkard corners with lifespan/deallocation. (But I don't have the build env to try.)

Gadgetoid, 2023-04-30:
If you want to have a crack at supporting fonts added via Python you can use GitHub to produce the builds for you. A little slow, but functional. I’d wanted to support this, and now I’ve got a pretty good idea how it might work since I’ve handled loading data from the filesystem in our JPEGDEC wrapper.

I think fixed-width is better handled in code, though, since it’s just a straight swap of the character width with the max character width and all bitmap fonts could support this.

Looks like there’s lots of empty byte bloat in the bitmap font data, too.

Gadgetoid, 2023-05-15:
Added a `fixed_width` kwarg to `measure_text` and `text` that forces bitmap fonts to use a fixed character width. You can grab an experimental build from the CI here: https://github.com/pimoroni/pimoroni-pico/pull/779

LionsPhil, 2023-06-22:
Sorry it took a while to get back to this, ended up trying the version in the 1.20.3 release, but it works! (Custom fonts are an awesome addition I'll have to try too.) Camera shutter even caught it mid-transition between 1 and other digits: :D

[image]

Leaving open because it needs doc; I'll see about adding a small PR for that.

closed 2023-06-23 by helgibbons

yjeanrenaud, 2023-09-14:
I would like to reopen this issue. The whitespace charcter is not monspaced. see:
[image]
[image]
I used ` badger.text(strLine, x, y, scale=2,spacing=1,fixed_width=True)` (yeah I know spacing is not a thing when using the bitmap fonts, but I tried to make sure.

Gadgetoid, 2023-09-14:
Ooof! 

https://github.com/pimoroni/pimoroni-pico/blob/36f3e3a69f411a5e5609dc437a75158d8f4787ef/libraries/bitmap_fonts/bitmap_fonts.cpp#L204

yjeanrenaud, 2023-09-14:
hmm.

Gadgetoid, 2023-09-14:
I've raised a PR to fix this, but for Badger 2040 builds to inherit this change they'll need a version bump.

PR #791 (2023-06-23): Document PicoGraphics fixed_width
(VSCode also stubbornly fixed some trailing space.)

Closes #765.
commit a0fe954b (2023-06-23): Document PicoGraphics fixed_width

(VSCode also stubbornly fixed some trailing space.)

Closes #765.