pimoroni/pimoroni-pico #766: Tufty w/ Firmware 1.20.1

insanire-monachus, 2023-05-03:
Tufty will not turn on after flashing it w/ firmware v.1.20.1 - revered back to 1.19.18.

(1) Flashed Tufty 2040 w/ 1.20.1
(2) System didn't boot nor turn on afterwards.
(3) Tried different USB ports on my laptop - nada
(4) Tried different USB cables - nada
(5) Bootloaded back to 1.19.18 - system booted, no issues.

Gadgetoid, 2023-05-03:
I can replicate this, though my local builds work fine. Trying to figure out why.

In the interim, v1.19.18 and v1.20.1 are basically the same.

insanire-monachus, 2023-05-09:
Curious, could you please explain further the details on how the Tufty firmware is overwriting itself? - Why has it not happened before, in the previously builds? - And why... there are missing checks and balances?

Always good to know why..

insanire-monachus, 2023-05-15:
Gadgetoid: Confirmed Tufty is back in action w/ 1.20.1 firmware. Awesome!! - Thanks so much!!  💯

insanire-monachus, 2023-05-15:
One item I noticed - previously to 1.20.2 firmware I was unable to run my badge under 565 due to memory error. But now it works. Though if I set the WHITE to 255, 255, 255 I get screen flickers (bad). So instead my WHITE is 246, 246, 247.

Gadgetoid, 2023-05-15:
Eek, that seems plausible @helgibbons - I don't know how that sailed by me. I can very easily replicate it but I don't have any great ideas why it's happening. Have moved over to that issue to explore further.

insanire-monachus, 2023-05-15:
"The flickering sounds like you might be running into https://github.com/pimoroni/pimoroni-pico/issues/567?" -- yes it is.

helgibbons, 2023-08-08:
Closing this one as I think the original issue is sorted.

closed 2023-08-08 by helgibbons