pimoroni/pimoroni-pico #770: How to compile micropython

VitRuzicka, 2023-05-08:
Hi, I need to make some changes in the source code of one of the libraries included in this project. How could I then compile your micropython  from source? Or is it possible to change pinout defined in the drivers->hub75.hpp in compiled version ?
Best regards,
Vit

VitRuzicka, 2023-05-09:
@Gadgetoid I would much rather prefer the fun way, so I'l just do a fork. Do I have to nudge somehow Github Actions to compile the code? Or does it work automatically?

Thanks,
Vit

ned-pcs, 2023-05-10:
I just had to figure this out myself.
Assuming you've checked out Micropython and pimoroni-pico repositories, you can do this:

```shell
MPY= # wherever you have the micropython repo
PMP= # wherever you have the pimoroni-pico repo
cd $MPY/ports/rp2
patch -p3 < $PMP/micropython/micropython_nano_specs.patch # only needs to be done once
make USER_C_MODULES=$PMP/micropython/modules/micropython-interstate75_ulab.cmake PICO_BUILD_DOCS=0 MICROPY_BOARD=PICO # add DEBUG=1 if debugging symbols are desired
# result will be in build-PICO/firmware.uf2
```

VitRuzicka, 2023-05-13:
Thank you @ned-pcs  I'll look into it.
VR

Gadgetoid, 2023-05-15:
This is more or less the build process from our CI for PICO:

```sh
git clone https://github.com/pimoroni/pimoroni-pico
git clone https://github.com/micropython/micropython

git -C pimoroni-pico submodule update --init

cd micropython/ports/rp2
make BOARD=PICO submodules

git apply ../../../pimoroni-pico/micropython/micropython_nano_specs.patch

cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DUSER_C_MODULES=../../../pimoroni-pico/micropython/modules/micropython-pico.cmake -DMICROPY_BOARD_DIR=`pwd`/../../../pimoroni-pico/micropython/board/PICO/ -DMICROPY_BOARD=pico
cmake --build build-PICO -j
```

And PICO_W:

```sh
git clone https://github.com/pimoroni/pimoroni-pico
git clone https://github.com/micropython/micropython

git -C pimoroni-pico submodule update --init

cd micropython/ports/rp2
make BOARD=PICO_W submodules

git apply ../../../pimoroni-pico/micropython/micropython_nano_specs.patch

cmake -S . -B build-PICO_W -DPICO_BUILD_DOCS=0 -DUSER_C_MODULES=../../../pimoroni-pico/micropython/modules/micropython-picow.cmake -DMICROPY_BOARD_DIR=`pwd`/../../../pimoroni-pico/micropython/board/PICO_W/ -DMICROPY_BOARD=pico_w
cmake --build build-PICO_W -j
```

This uses `-DMICROPY_BOARD_DIR` (a CMake arg) to specify our version of the `PICO_W` board dir, which ensures various libs like `gfx_pack.py`, `pimoroni.py` and our `boot.py` are included. Without `boot.py` - which switches our C++ memory shim over to using MicroPython's heap - you may run into C++ OOM issues.

Some boards - like Inky - also have a `pico_sdk.patch` which add some startup clock config, RTC config and GPIO latching. If you see a build error like `undefined reference to `runtime_wakeup_gpio_state'` then you're missing that patch.

Details of which `BOARD`, CMake file and whether a board-specific patch is needed can be gleaned from the CI config: https://github.com/pimoroni/pimoroni-pico/blob/00d161794736d545d933181566409c5b2cc3a8dc/.github/workflows/micropython.yml#L69-L90

Obsttube, 2023-11-27:
If anyone has problems with that then remember that PICO got renamed to RPI_PICO and PICO_W to RPI_PICO_W