pimoroni/pimoroni-pico #772: How are images to be loaded onto the Badger2040w?

KnownBySomeone, 2023-05-09:
I have not yet figured out how to load image files onto the Badger2040w and get them to display.

Steps I have taken:
-I read the readme within /examples/image.py """"Images must be 296x128 pixel JPEGs   Create a new "images" directory via Thonny, and upload your .jpg files there."""

-I created what I believe to be a 1bit, 296x128 pixel .jpg image file.  The same file as attached.

-I loaded said file into the /images folder alongside the /images/badgerpunk.jpg file, and got the message, "JPEG: could not read file/buffer. JPEG: could not read file/buffer."

-I have also loaded the image in other locations that did not follow the readme, but have had no success as well.

-I have also separately tried changing which extra data is saved and tried saving the image as grayscale when extracting the image from GIMP.

[image]

Gadgetoid, 2023-05-09:
What's the code you're using to attempt to load the image?

Does a regular `open("/images/filename.jpg")` work?

KnownBySomeone, 2023-05-09:
As instructed in the readme, and as I am used to using with MicroPython, I loaded the image file using the graphical interface included in Thonny.

> What's the code you're using to attempt to load the image?
> 
> Does a regular `open("/images/filename.jpg")` work?

[image]

Gadgetoid, 2023-05-09:
That `open` line is Python code. For example: `open("/images/badgerpunk.jpg")` should work in your current setup.

After uploading the file, so should `open("/images/CatTree01.jpg")` and using the JPEG decoder: `jpeg.open_file("/images/CatTree01.jpg")`

KnownBySomeone, 2023-05-10:
I recreated the steps I took in this video.

https://youtu.be/Q-kZRGcYm40

helgibbons, 2023-06-15:
If you're still having trouble - it might be worth checking that you're saving your .jpgs as non progressive? The image you posted above didn't work for me, but this one (opened in GIMP and exported without the progressive flag ticked) seems to work OK:

[image]