pimoroni/pimoroni-pico #782: Pico Graphics: Can't set thickness

nicosuave, 2023-05-21:
I'm using an Inky 7 frame and seeing `RuntimeError: Thickness not supported!` when calling `set_thickness()` on an instance. Am I possibly behind a version or two?

andrewrump, 2023-05-21:
`set_thickness()` is not support anymore (or currently) but I have implemented a similar looking result by just calling the drawing functions, i.e., `text()` several times offsetting the x and y one pixel for each call to obtain the same results depending on the .thickness wanted.

helgibbons, 2023-05-22:
I just tried `set_thickness()` with the most recent MicroPython version on Inky and it seems to be working?

nicosuave, 2023-05-22:
Thanks @helgibbons my board must be out of date.

> `set_thickness()` is not support anymore (or currently) but I have implemented a similar looking result by just calling the drawing functions, i.e., `text()` several times offsetting the x and y one pixel for each call to obtain the same results depending on the .thickness wanted.

This worked surprisingly well!

closed 2023-05-22 by nicosuave