pimoroni/pimoroni-pico #787: Bluetooth integration into MicroPython

tomlahr, 2023-06-15:
Is the new Bluetooth functionality coming in the foreseeable future
into a new MicroPython Build?

TiA.

alphanumeric007, 2023-06-15:
Looks like its being worked on.
https://github.com/pimoroni/pimoroni-pico/pull/716

DuaneKaufman, 2023-06-19:
Dear All,
I have a Badger Pico 2040W upon which I added the latest FW (with bluetooth):

MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, picow v1.20.3 on 2023-06-19; Raspberry Pi Pico W with RP2040

but when I try to run launcher.py (from Thonny), I get the error:

%Run -c $EDITOR_CONTENT
Traceback (most recent call last):
File "", line 1, in
File "launcher.py", line 5, in
File "/lib/badger2040w.py", line 10, in
ImportError: no module named 'wakeup'

Your help is appreciated,
Duane

Gadgetoid, 2023-06-20:
There is no Bluetooth firmware for Badger 2040W yet...

North101, 2023-06-20:
I used these micropython examples to test bluetooth on the plasma stick and it was super quick to get up and running. I even wrote a flutter app the communicate with it
https://github.com/micropython/micropython/tree/master/examples/bluetooth

I have made a PR to slightly improve `bluetooth_advertising.py` here:
https://github.com/micropython/micropython/pull/11831

DuaneKaufman, 2023-06-20:
> There is no Bluetooth firmware for Badger 2040W yet...

Oh, I took this:
https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.20.3

to mean - "This release bumps the MicroPython version we're building against from 1.20 to 856e08b1931b88271816a2f60648f6ff332235b2 and includes various Bluetooth-related libraries into our build. Bluetooth support should be at feature parity with the official MicroPython release."

I appreciate all the effort the team is putting into this.
Sincerely,
Duane

Gadgetoid, 2023-06-20:
Ah badger builds now live in the Badger2040 repository, and there's a pending PR for Bluetooth here which I need to fix - https://github.com/pimoroni/badger2040/pull/28

North101, 2023-06-28:
When using one picow as a central, can only 1 peripheral be connected at a time?
Connecting to the 1st device works fine, but connecting to any further devices causes the following error:
```
OSError: [Errno 107] ENOTCONN
```

Gadgetoid, 2023-06-28:
I see the same thing trying to connect to multiple Enviro sensors:

```
Connecting to Device(ADDR_PUBLIC, 28:cd:c1:00:79:3e)
Connecting to Device(ADDR_PUBLIC, 28:cd:c1:00:78:5c)
Traceback (most recent call last):
  File "<stdin>", line 187, in <module>
  File "asyncio/core.py", line 1, in run
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 184, in main
  File "asyncio/funcs.py", line 1, in gather
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 109, in client_task
  File "aioble/device.py", line 142, in connect
  File "aioble/central.py", line 125, in _connect
OSError: [Errno 107] ENOTCONN
```

I wonder if it's necessary to round-robin the devices... I don't have a good answer to this since I'm still figuring it out myself.

Streaming readings always seems to hit a 60sec connection timeout for me anyway, throwing an `AttributeError`, `TypeError`, `aioble.DeviceDisconnectedError`, `aioble.GattError` or `asyncio.TimeoutError` depending on when exactly the connection is dropped.

Gadgetoid, 2023-06-29:
Ooof. Round-robin it is then! Though I am rather curious what might happen if I bumped that up to 2 or more.

Gadgetoid, 2024-02-27:
Closing as stale/completed.