pimoroni/pimoroni-pico #798: Word Clock on Inky Frame - UTC only!

wholepunch, 2023-07-09:
Hey. I cannot, for the life of me, work out how to allow for timezone in the world clock. I've tried a few bodges, but nothing sticks for long and the time drifts out again. I'm in GB (UTC +1 atm) - how the heck do I reflect that in the code? Thanks!

Gadgetoid, 2023-07-10:
MicroPython has no timezone support so you've got to add an hour to the time as it's ready back from the RTC.

Probably somewhere around here:

https://github.com/pimoroni/pimoroni-pico/blob/25237c54ce465279134fe429e4c11ffe5cb01cfd/micropython/examples/inky_frame/inkylauncher/word_clock.py#L42C31-L42C31

And dealing with rollover is a nightmare, so you might need to convert into a UNIX timecode and add an hour in seconds, which is sort of what's happening here:

https://github.com/ahnlak/badger2040/blob/bc1a666f4042c1018fe40faaf94ab2f4e6086dba/firmware/PIMORONI_BADGER2040W/lib/badger2040.py#L145-L151

wholepunch, 2023-07-10:
Ok, thanks, iI'll have a punt at inserting the additional hour where you suggest.

wholepunch, 2023-07-10:
Think I got it. 

hours = hours + 1 

here https://github.com/pimoroni/pimoroni-pico/blob/25237c54ce465279134fe429e4c11ffe5cb01cfd/micropython/examples/inky_frame/inkylauncher/word_clock.py#L19

Obv I'll need to update it for the clock changes, but it's easier than trying to get epoch time working (for me, anyway..)

Thanks!

North101, 2023-07-12:
This is how i'm adding a utc offset:
```python
import utime

def get_time(utc_offset: int):
  # utc_offset is in seconds
  utime.localtime(utime.mktime(utime.localtime()) + utc_offset)
```

However just setting a static offset doesn't work because of daylight savings. So you could use https://ip-api.com/ to lookup the offset based on your IP.
```python
import urequest
# limited to 45 / minute but really you should only do it once a day
def get_utc_offset():
  r = urequest.get('http://ip-api.com/json/?fields=offset')
  return r.json()['offset']
```

thirdr, 2024-05-30:
Issue appears to have been resolved. Please reopen if required :)

closed 2024-05-30 by thirdr