pimoroni/pimoroni-pico #801: Inky frame sleep

exussum12, 2023-07-18:
The manual says

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules_py/inky_frame.md#rtc

I get 
`AttributeError: 'module' object has no attribute 'sleep_for'`

when running that code

The example code seems to work 

```
import inky_helper as ih
ih.sleep(minutes)
```

Do I need to update my frame? or example code need updating to follow docs?

helgibbons, 2023-07-19:
`sleep_for` was a reasonably recent addition I believe - are you using the most the recent Inky Frame MicroPython build from the [releases page](https://github.com/pimoroni/pimoroni-pico/releases)?

exussum12, 2023-07-22:
I only received it the day before this, assumed it was on latest version. Its mounted on a wall now so next time I update code I will flash and verify

iohe, 2023-08-03:
Same happened for me. Received Inky Frame 7.3 board today and after writing pimoroni-inky_frame-v1.20.3-micropython.uf2 issue was solved. Thank you very much!

helgibbons, 2023-08-03:
Good stuff - I'll check with production what version we're preloading on new units and get them to bump it to the newest version if necessary.

exussum12, 2023-08-04:
Confirmed it was an old version too. Updating resolved

daniel-j, 2023-08-21:
Did you get sleeping with timer working? I can't get it to work on my Inky Frame 5.7 or 7.3 on battery. It either never wakes up or wakes up instantly. Tested both latest MP release and the C++ examples.

exussum12, 2023-08-21:
Yeah it works fine for mine. Assume your powering with the JST connection not the USB?

thirdr, 2024-06-03:
Issue appears to have been resolved. Closing, please reopen if required :)

closed 2024-06-03 by thirdr