pimoroni/pimoroni-pico #805: Interstate 75 W, no global brightness function?

alphanumeric007, 2023-07-26:
Interstate 75 W, Pimoroni Micro python uf2 V120.3 

I was wanting to adjust the brightness of my array based on background light. Hunting around through the documentation for the Interstate 75 I noticed there wasn't any global brightness function. I eventually got it done with **hsv**. It works but IMHO, its not an ideal way to do it. I had to be pointed to the color wheel to figure out the color values to enter.

If this feature can be added, it will save the next guy a lot of time and maybe some frustration. 

https://www.cssscript.com/demo/hsv-hsl-color-wheel-picker-reinvented/
https://forums.pimoroni.com/t/interstate-74-w-how-do-you-control-the-display-brightness/22591/14

North101, 2023-07-28:
It would be great to have the unicorn's set brightness function ported to the interstate75 and the plasma libraries

alphanumeric007, 2023-07-28:
@Gadgetoid, thanks for taking the time to take a look. 
I've often found that , what I think would likely be an easy fix, isn't.
@helgibbons really helped me out on this one, even after my whining. 
So, I have a working solution, and no rush. =)

helgibbons, 2023-07-31:
Going to leave this one open as a reminder :)

whatleymd, 2024-06-09:
+1 for having a global brightness function for I75. Text is doable per the HSV workaround, but images cause temporary vision loss. I'm handling images via hub75 micropython library and painting the image via set_pixel. If you can provide a high level summary of necessary changes required in the c++ library, maybe someone can come along with a PR? Thanks

Gadgetoid, 2025-04-25:
Linekd from https://github.com/pimoroni/interstate75/issues/32 to keep this on the deck, but closing here since the Interstate75 repo is the new home for the project.

My working theory is to make the gamma table mutable and provide functions to adjust it. It wont offer a huge range of values, but it should be pretty much cost neutral if you're not constantly updating the brightness.

closed 2025-04-25 by Gadgetoid