pimoroni/pimoroni-pico #814: main.py not loading.

fableman73, 2023-08-10:
Problem: Everything works perfect with usb cable inserted and I run my python program with the screen and all.
Then I remove the USB cable and hold BOOTSEL and then transfer my program to the pico flash and name it main.py
Then I remove the USB cable and insert into a USB power supply.
Nothing hapends.   Then main.py dos not run. I even tried the most simple LED blink program, still not loading.

I tried the flash_nuke.uf2 to clear all and reload latest firmware. pimoroni-picow-v1.20.4-micropython.uf2

Still no luck,,, problem is that I can't test without the screen because i soldered some stuff.
I read up on all guides I can find and now I start to believe it have something todo with your firmware or the screen connected it self..?

any idees?  Could someone try, if it working for you to run any program as main.py connected without computer and USB.

helgibbons, 2023-08-11:
You only need to use bootsel/DFU mode to copy the MicroPython firmware to the board. You should be saving your `main.py` to the board using Thonny - there's a tutorial here that might help: https://learn.pimoroni.com/article/getting-started-with-pico#using-thonny-hello-world-and-blink

fableman73, 2023-08-11:
No that is wrong.. I found the problem (the next day) after 5 hours banging
my head to the desk :)
A lot of wrong information is out there I can tell...

You must transfer the main.py using ampy

Problem: The micropython is not sharing the same flash area that BOOTSEL
mount after pressing it. So the files you put there can't be seen by
micropython.

Solution: transfer files using ampy

https://github.com/scientifichackers/ampy

helgibbons, 2023-08-11:
Good stuff, glad you figured it out :)

There's lots more tools around for working with RP2040 flavoured MicroPython than there used to be - I've been recently enjoying using the [MicroPico plug-in](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go) for developing from Visual Studio Code.

closed 2023-08-11 by helgibbons