pimoroni/pimoroni-pico #820: Pimoroni Encoder i2c Breakout not working correctly when going clockwise?

alphanumeric007, 2023-08-16:
When running the demo.py file for the Breakout Garden i2c encoder it's not registering the knob position correctly.
Going from position 5 to 6, knob at 3 o'clock, it doesn't detect the change. Move it one more and it updates to 6, it's in position 7 though. There are other similar glitches on the way to position 24, 12 o'clock high.
If I go CCW from 24 to 0 it works just fine though?
And  `enc.set_direction(BreakoutEncoder.DIRECTION_CCW)`  makes no difference.
Clockwise it glitches, CCW its OK.

alphanumeric007, 2023-08-17:
Should have mentioned. It's wired (soldered) to an Inventor 2040 W BG I2C pads.
The Inventor 2040W is running the PicoW,120.4 uf2.

alphanumeric007, 2023-08-17:
Same deal if I use machine i2c. I'm thinking its a hardware issue with the Encoder?
```
import machine
sda=machine.Pin(4) # Explorer 20 Breakout 4
scl=machine.Pin(5) # Explorer 21 Breakout 5
i2c=machine.I2C(0,sda=sda, scl=scl, freq=400000)
 

#from pimoroni_i2c import PimoroniI2C
from breakout_encoder import BreakoutEncoder
#i2c = PimoroniI2C(sda=(4), scl=(5))
```

helgibbons, 2023-08-18:
We've not been able to replicate this behaviour and positions 6 / 24 both seem present , correct and behaving as expected on our encoder so I suspect you might be right...

alphanumeric007, 2023-08-18:
Yeah, just weird that it works CCW but not CW? Time to warm up the soldering iron.

closed 2023-08-18 by alphanumeric007