pimoroni/pimoroni-pico #823: Not enough memory for PNG decoding on Inky Frame 5.7"

lrasinen, 2023-08-24:
I'm trying to do a weather display with 50x50 weather icons, but creating the `pngdec.PNG` object fails because it wants the 48k of memory right off the bat.

As @bitbank2 commented (https://github.com/pimoroni/pimoroni-pico/pull/802#issuecomment-1645440731) I don't need the full 32k sliding window for my tiny icons. If that was configurable during initialization it'd be great.

bitbank2, 2023-08-24:
I understand the situation and would like to assist, but I'm super busy on other projects. I'm open to proposals to change the code to allow flexible sliding window sizes at runtime. Could you propose the needed changes?

lrasinen, 2023-08-24:
Had a quick look through the code and the sliding window buffer is allocated statically: https://github.com/pimoroni/pimoroni-pico/blob/57042bfed3b2fbd45b2e1c36582e3b9bfff3faf9/libraries/pngdec/PNGdec.h#L152C5-L152C48

If that could be given as a constructor parameter and then add all the necessary plumbing back to the Python level that'd probably do it.