pimoroni/pimoroni-pico #827: Galactic Unicorn scrolling text example

exussum12, 2023-08-29:
I'm trying to display a very large amount of scrolling text on the unicorn, the example doesn't work with it due to it basically rendering the whole string each time.

I have made some improvements to only render 20 chars at a time, which now makes the display much more responsive.

is this something wanted in the examples? Realise it may be a niche case

Gadgetoid, 2023-08-29:
I'd be interested to see your approach, at the very least. A clear demo of the problem could give me some hints for how to tackle it at the PicoGraphics level.

exussum12, 2023-08-29:
Changes before the loop

```
MESSAGE = "Paste lyrics from a reasonable length song"
max_display = 20
size_to_chunk_at = 150
partial_text = 0
if (len(message_display) > size_to_chunk_at):
    message_display = MESSAGE[0:max_display]
    current_char = 0
    partal_text = 1
msg_width = graphics.measure_text(message_display, 1)
while True:
```

Changes inside the loop
```

    if (partial_text = 1 and PADDING - shift < -10):
        char_size = graphics.measure_text(message_display[0:1], 1)
        if (message_display[0:1] == ' '):
            char_size -= 1 # Not sure why this is needed, feels like a bug somewhere
        shift -= char_size
        current_char += 1
        message_display = MESSAGE[current_char:current_char + max_display]
    graphics.clear() # Existing line

    outline_text(message_display, x=PADDING - shift, y=2) # changed to use new variable
```

so a new few new variables, one to track the current index of the start of the text being displayed, one to track either the whole string or the partial string being displayed, and one to track if the partial display method is being switched on avoid it for the smaller strings as it doesn't make any real difference there

At the pico graphics level Im guessing it would need to be approached with the size of each letter pre cached, and instead of rendering the pixels, just summing the values so if offset was -15, you would loop over the string one at a time, eg string = ABCDEFG, A = 5, remove from offset and dont render, B = 5, remove from offset and dont render, C = 4, Remove from offset and dont render. D = 5, this now straggles offset, so render string DEF from offset -1.

End of the string would be similar.

As a user I would normally expect this to be user side though, as there are advantages to pre rendering, can re use the memory for example and have no processing