pimoroni/pimoroni-pico #849: SD card timeout errors with sdcard.mpy

helgibbons, 2023-09-20:
I was getting timeout errors (as per https://github.com/micropython/micropython/issues/7129 ) with certain (Samsung) SD cards with the compiled sdcard.mpy in micropython/examples/common. These were resolved when I switched to [the most recent version of the library](https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py) - I think our compiled version might need updating?

Gadgetoid, 2023-09-21:
I think we need to delete those files altogether and do this - https://github.com/pimoroni/picovision/pull/41 - which might require some documentation changes.

Gadgetoid, 2023-10-06:
Added this to https://github.com/pimoroni/pimoroni-pico/pull/858

closed 2024-05-31 by thirdr