pimoroni/pimoroni-pico #851: ImportError: can't import name PEN_3BIT

ptvagner, 2023-09-24:
Using the latest Inky Frame firmware:

> MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, inky_frame v1.20.6 on 2023-09-18; Raspberry Pi Pico W with RP2040

importing:

`from picographics import PicoGraphics, PEN_3BIT, DISPLAY_INKY_FRAME_7 as DISPLAY  # 7.3"`

error:

```
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
ImportError: can't import name PEN_3BIT
```

according to the documentation this should be a valid mode.

[image]

Gadgetoid, 2023-09-24:
There's no need to specify a pen type with Inky Frame, they are forced to specific values since nothing else is supported.

Inky frame 7" doesn't actually use 3-bit at all, since it needs a special pen type to handle the PSRAM and- since the PSRAM is *huge* actually just does native 4-bit internally.

We should probably clarify these exceptions in the docs.

ptvagner, 2023-09-25:
@Gadgetoid 

> There's no need to specify a pen type with Inky Frame, they are forced to specific values since nothing else is supported.

aha! that makes sense, thanks for clearing that up :)

closed 2023-09-25 by ptvagner