pimoroni/pimoroni-pico #854: request: add deflate/zlib/gzip compression support

grunkyb, 2023-09-26:
The introduction of the [`deflate`](https://docs.micropython.org/en/latest/library/deflate.html#deflate.DeflateIO) module is slated for Micropython 1.21, but that release is months overdue. Any chance that module can be added to a pirate-flavour release? Specifically, I want to use `zlib.compress`, but 1.20.x only includes the `decompress` function and `DecompIO` class.

Gadgetoid, 2023-10-05:
I've not had much luck or experience generating anything but super trivial .mpy files.

It looks like Deflate was merged in https://github.com/micropython/micropython/commit/3533924c36ae85ce6e8bf8598dd71cf16bbdb10b and we're already targeting a pre-release commit of MicroPython, so I just need to find the time to walk it forward to the latest upstream commit and fix all the breaking changes. (There have been some renames to MicroPython RP2 boards which will break out build.)

Gadgetoid, 2023-10-05:
Okay apparently that was not as difficult as I'd thought. The "deflate" module should be in these builds: https://github.com/pimoroni/pimoroni-pico/actions/runs/6418396733?pr=858

grunkyb, 2023-10-05:
Thanks @Gadgetoid
At some point I want to learn what got changed in that commit to allow it to be built.

closed 2023-10-05 by grunkyb

Gadgetoid, 2023-10-06:
Wow... https://github.com/micropython/micropython/releases/tag/v1.21.0 just released :laughing: 

I think we were accidentally prescient here.

I think our build worked pretty much fine without changes by just bumping the commit has to the latest MicroPython, but no doubt the various changes will have far reaching repercussions beyond "does it build."

grunkyb, 2023-10-10:
For anyone else finding this thread who wants to enable compression with [`deflate.DeflateIO`](https://docs.micropython.org/en/latest/library/deflate.html#module-deflate), I recommend as a starting point [this tutorial on Medium](https://medium.com/@iestynlloyd/galactic-unicorns-and-custom-pimoroni-pico-firmware-38dd7c5913b8) and [this post from @Gadgetoid](https://github.com/pimoroni/pimoroni-pico/issues/286#issuecomment-1063821276). In my case, I added the line `#define MICROPY_PY_DEFLATE_COMPRESS (1)` to `mpconfigboard.h` for the Pico W. The file is in `micropython/board/RPI_PICO_W/`. The actions take care of the rest of the firmware build.