pimoroni/pimoroni-pico #856: VL53L5CX on Pico W stuck at "Starting up sensor..."

slabua, 2023-10-03:
I seem to have no luck in running the vl53l5cx demos on the Pico W, keep getting stuck at
```
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
Starting up sensor...
```
I have tried different releases, also nuked the pico a few times, but still have the same problem.

If I press Stop, after a while I get this error:
```
Could not interrupt current process. Please wait, try again or select Stop/Restart!

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialTimeoutException: Write timeout).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
```
Once the Pico is in this situation, I have to unplug it and plug it back to be able to access the repl.

helgibbons, 2023-10-06:
Have you uploaded the firmware blob to your Pico?

# The VL53L5CX requires a firmware blob to start up.
# Make sure you upload "vl53l5cx_firmware.bin" via Thonny to the root of your filesystem
# You can find it here: https://github.com/ST-mirror/VL53L5CX_ULD_driver/blob/no-fw/lite/en/vl53l5cx_firmware.bin

Step by step instructions are here: https://learn.pimoroni.com/article/micropython-and-vl53l5cx

slabua, 2023-10-06:
> Have you uploaded the firmware blob to your Pico?
@helgibbons yes I did. (In the case of missing firmware, the error is different).
Should I try a particular version of micropython, other than the latest and the one specified in the instructions?

Gadgetoid, 2023-10-06:
A quick test with the latest MicroPython build from https://github.com/pimoroni/pimoroni-pico/pull/858

Running from a Pico W Explorer using a Qw'St to Qw'St cable to ensure a good connection, and I've got numbers pouring out.

What's your wiring setup?

Looks like incorrect i2c pins produce:

```python
RuntimeError: VL53L5CX: init error
```

Which is *not* what's happening for you, but I thought I'd check just in ccccase.

helgibbons, 2023-10-06:
Working OK for me too (with vl53l5cx_demo.py and v1.20.6).

slabua, 2023-10-06:
@Gadgetoid @helgibbons
I thought to check the md5sum of the binary blob I had, and the one I just downloaded anew for verification~
The non working version had md5sum `f396addf82ce7abc6dd56f3997164d5a` whilst the working version has `0f6e20ae12db3ec6a37cf3adcb46320a`.
Not sure why, perhaps the first time it got truncated because the Pico storage was full!
It is working properly now, thank you!

thirdr, 2025-01-08:
Closed as resolved.

closed 2025-01-08 by thirdr