pimoroni/pimoroni-pico #857: Servo2040: servos 1 and 17 and 2 and 18 moving together

firkowski, 2023-10-05:
hello, I am working on a hexapod project and i am using Servo2040 board with micro-ROS. I noticed strange behavior when running my C++ code. servos 1 and 17 would shake uncontrollably and change directions rapidly. it took me a while to realize that servos 1 and 17 and 2 and 18 were moving in the same way. I may have an issue with the I2C causing them to receive the same signals (i think the reason for the shaking was that servos 1 and 17 were receiving two signals at once). I would like to be able to control them independently. My full code (modified for testing) is below but I've highlighted the part that causes problems.

problematic part:
```
void init() {
	for (int i = 0; i < NUM_SERVOS; i++) {
		cals[i] = std::make_unique<Calibration>();
		cals[i]->apply_three_pairs(min_pulses[i]      , mid_pulses[i], max_pulses[i]     ,
                               -WIDE_ANGLE_RANGE/2,             0, WIDE_ANGLE_RANGE/2);
		servos[i] = std::make_unique<Servo>(i, *cals[i]);
		servos[i]->init();
	}
	for (int i = 0; i < NUM_SERVOS; i++) {
		servos[i]->enable();
	}
	servos[16]->value(-30); //causes servo 16 and 1 to move
	servos[17]->value(45); //causes servo 17 and 2 to move**

```

whole code:
```
#include <stdio.h>
#include "pico/stdlib.h"

#include "rclc/rclc.h"

#include <time.h>
#include <vector>
#include <memory>
#include <map>
#include <cmath>

#include "servo2040_uros/msg/feet_contacts.h"
#include "servo2040_uros/msg/current.h"
#include "servo2040_uros/msg/joint_angles.h"
#include "sensor_msgs/msg/joint_state.h"

extern "C"{
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include "pico_uart_transports.h"
}

#include "servo2040.hpp"
#include "analogmux.hpp"
#include "analog.hpp"
#include "button.hpp"

#define NUM_LEGS 6
#define NUM_SERVOS_PER_LEG 3
#define NUM_SERVOS 18

#define SERVO_L1C 0 //1
#define SERVO_L1F 1 //2
#define SERVO_L1T 2 //3
#define SERVO_L2C 3 //4
#define SERVO_L2F 4 //5
#define SERVO_L2T 5 //6
#define SERVO_L3C 6 //7
#define SERVO_L3F 7 //8
#define SERVO_L3T 8 //9
#define SERVO_R1C 9 //16
#define SERVO_R1F 10 //17
#define SERVO_R1T 11 //18
#define SERVO_R2C 12 //13 
[... 241 lines cut ...]
    	switch (state) {
    	    case WAITING_AGENT:
    	      state = pingAgent() ? AGENT_AVAILABLE : WAITING_AGENT;
    	      break;
    	    case AGENT_AVAILABLE:
    	      createEntities();
    	      state = AGENT_CONNECTED ;
    	      break;
    	    case AGENT_CONNECTED:
    	      state = pingAgent() ? AGENT_CONNECTED : AGENT_DISCONNECTED;
    	      if (state == AGENT_CONNECTED) {
    	        rclc_executor_spin(&executor);
    	      }
    	      break;
    	    case AGENT_DISCONNECTED:
    	      destroyEntities();
    	      state = WAITING_AGENT;
    	      break;
    	    default:
    	      break;
    	  }
    }
}

```

ZodiusInfuser, 2023-10-05:
Hi @firkowski,

Cool to see someone using Servo 2040 for the primary purpose I designed it for!

The issue you are experiencing is expected, due to the RP2040 *only having 16 PWM channels. As such when you create the 17th and 18th servo classes, even though they are independent classes, the same hardware registers will be controlled, resulting in what you describe.

The way around this is to switch from using the `Servo` class to using the `ServoCluster` class. This is a PIO implementation of PWM that supports 1 to 30 of RP2040's pins. It also offers phase control, so that servos don't all receive their commands at the same time, spreading out the peak current draw.

You can find the code for `ServoCluster` at: https://github.com/pimoroni/pimoroni-pico/blob/main/drivers/servo/servo_cluster.hpp

It seems I never got to writing documentation for the C implementation of this class, but the Micropython documentation should help you. https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/servo#servocluster
There is also this example: https://github.com/pimoroni/pimoroni-pico/blob/main/examples/servo2040/servo2040_servo_cluster.cpp

This should be a relatively painless transition for your code, as the APIs for the two classes were written concurrently (they both use the same `ServoState` object internally). So for your code, the main change would be to swap `std::array<std::unique_ptr<Servo>, NUM_SERVOS> servos;` over to a single `ServoCluster` instance.

Hope that helps. Let me know if you have any specific questions

ZodiusInfuser, 2023-10-07:
Closing this. If there are any bugs with getting it working, please reopen this or raise a new issue.

closed 2023-10-07 by ZodiusInfuser

firkowski, 2023-10-14:
It works great now, thank you! One thing is that I can't figure out how to apply calibration to the servos in ServoCluster. There does not seem to be a calibration(servo, calibration) function in the c++ code, or at least i can't find it