pimoroni/pimoroni-pico #859: Brandnew 2040W has newer FW/examples than V0.0.3 FW here

Arthur-Philip-Dent, 2023-10-08:
It looks like you did the "pimoroni-badger2040-v0.0.3-micropython-with-badger-os.uf2" and the "pimoroni-badger2040w-v0.0.3-micropython-with-badger-os.uf2" with the 2040 firmware, since it the "pimoroni-badger2040w-v0.0.3-micropython-with-badger-os.uf2" does not know either modul 2040w nor badger_os, which is used in the examples of 
[Link2040W](https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/badger2040w). 

I ended up flashing the "pimoroni-badger2040w-v0.0.3-micropython-with-badger-os.uf2" and all of my backed up example-files didn't run anymore. Took the [.uf2](https://github.com/pimoroni/badger2040/releases/download/v0.0.3/pimoroni-badger2040w-v0.0.3-micropython-with-badger-os.uf2) with an 7 months old version of the badge_os and the brandnew delivered badger doesnt run anymore. Can you please supply the new files? Thanks!

exussum12, 2023-10-09:
It has moved here
https://github.com/pimoroni/badger2040
Releases here

https://github.com/pimoroni/badger2040/releases/tag/v0.0.3

Arthur-Philip-Dent, 2023-10-09:
Thanks for reply, @exussum12!
Thats exactly where I went and nothing works anymore. 
2030w Modul not known.
And when I install the 2030w *.uf2 package, WiFi doesn’t work (clock.py). 
Also badger_os.warning (in the newer examples) is not know and causes the *.py not running. 

The newer code does not fit to the V0.0.3 *.uf2 fw , it seems. 

Seems lately many people have this problem only with 2030w, so I guess

Arthur-Philip-Dent, 2023-10-09:
Was there. Done that. 
I think you are on it. 

Take a 2040w and do like you proposed. 
And see, if I‘m wrong. 

The (new) code from the given link doesn’t run with the V0.0.3 firmware. 
I installed the firmware V0.0.3 with Badger_OS all fine, I can start any *py. 
The 2040w V0.0.3 code has no module 2040w for import… 
If I install the linked examples of September 2023, they won’t work. 
They import 2040w, which does not come with the firmware 2040w nor with the firmware2040w-badger_os.

The V0.0.3 firmware for the 2040w doesn’t know the modul 2040w.

Gadgetoid, 2023-10-09:
There is not, and never has been a `2040w` module. If you're trying to `import badger2040w` it's because you're running old code on a new firmware. It should be `import badger2040` so code is portable between 2040 and 2040W.

I flashed the mentioned firmware to my Badger 2040W and the examples run fine.

The docs and examples in this repository are out of date and should probably be deleted now.

* :x: Out of date: https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/badger2040w
* :heavy_check_mark:  Correct examples: https://github.com/pimoroni/badger2040/tree/main/badger_os/examples

Arthur-Philip-Dent, 2023-10-09:
@Gadgetoid: oh dang! 😝
I tried to ride old horses and actually did everything right - except for sticking my nose it files from September 2023. Holy crap! 🫣

Thanks 🙏🏻 for clarifying!

closed 2023-10-09 by Arthur-Philip-Dent