pimoroni/pimoroni-pico #863: How to apply custom calibration to ServoCluster in C/C++

firkowski, 2023-10-17:
Hi, I'm using C/C++ ServoCluster to control 18 servos on Servo2040 but I do not know how to apply custom calibration to it. There does not seem to be a ServoCluster::calibration(<servo>, <calibration>) function like in the Micropython library, or at least I can't find it.

ZodiusInfuser, 2023-10-18:
Hi @firkowski,

Accessing the calibration in C++ is slightly different to Micropython. It is a function that returns a reference to the calibration, letting you modify it directly. Here are the functions:

https://github.com/pimoroni/pimoroni-pico/blob/b4451c3bdc06235a1358a5a8aabd008647ed9f8a/drivers/servo/servo_cluster.cpp#L482-L490

And here is a usage example with the regular C++ Servo class: https://github.com/pimoroni/pimoroni-pico/blob/main/examples/servo2040/servo2040_calibration.cpp

P.S I think you are the first person other than myself who's used the calibration feature (or that at least has asked about it)!

Let me know how you get on :)

firkowski, 2023-10-18:
Thank you @ZodiusInfuser, this is exactly what i needed :) It works flawlessly.

I'm pretty sure @MakeYourPet has used calibration extensively in his hexapod. I hope you are aware of his project because it's very impressive and has gathered a sizable community around his design

closed 2023-10-18 by firkowski

ZodiusInfuser, 2023-10-18:
Glad that worked!

You mentioned that project in your last Servo 2040 issue, but I only checked out their YouTube channel, not their github so missed the very obvious diagram on the main page 😅