pimoroni/pimoroni-pico #865: Functions for displaying PNG files are missing from PicoGraphics readmes

helgibbons, 2023-10-23:
They can mostly be deducted from the JPG functions, but they would be less of a mystery if we added them to the relevant readmes :)

barchard, 2023-12-26:
I can't seem to get `import pngdec` to work like `import jpegdec` does. What did you use for your `import` statement? Thanks

helgibbons, 2024-01-03:
@barchard This works for me on a Tufty 2040. I think `pngdec` was quite a recent addition, so it might be worth [downloading the most MicroPython image](https://github.com/pimoroni/pimoroni-pico/releases) for your board if you're having trouble?

```
import picographics
import pngdec

display = picographics.PicoGraphics(display=picographics.DISPLAY_TUFTY_2040)

# Create a new PNG decoder for our PicoGraphics
p = pngdec.PNG(display)

# Open the PNG file
p.open_file("filename.png")

# Decode the PNG
p.decode(0, 0)

# Display the result
display.update()
```

thirdr, 2024-05-30:
The README [here](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/modules/picographics/README.md#png-files) has been updated now :).

closed 2024-05-30 by thirdr