pimoroni/pimoroni-pico #866: Update of graphics.update() seems to get stuck indefinatelly

klausi83, 2023-10-25:
I have written a py-script that should convert inky_frame to a small self contained picture frame. (Inky_frame 7.3). Unfortunately I get an unexplained hang-up.
Code: https://github.com/klausi83/inky_frame
The code seems to run ok, for one or two pictures after first initialization (coming from main.py and selecting Button "C"). The event inky_frame.woken_by_rtc() triggers but almost always, the update of the displayed picture gets stuck on line 124 and 127 (most likely in line 127). A press of the "reset" usually does not help, it gets stuck at the same point.

Getting out of this status is: press reset + (A and E) and start from main.py again.

Is there a way to debug this deeper, or does someone know a solution?

klausi83, 2023-11-04:
I tested some more today. Tried twofold
1. Increasing update frequency to 2 Minutes. Running it via USB and Thonny for more than an hour. All graphics-updates worked like they are supposed to.
2. Increased update frequency to 2 Minutes. Running without USB connection on a variable power supply. Tried everything between 2,8V and 5,4V. In all cases the controller and the display worked. Running the code yielded the above error every trial run after only a few graphics updates.
So, to me, there seems to be some mayor difference in graphics.update() routine depending on inky-frame running in USB-Mode or running in stand-alone mode. Is someone more knowledgeable than me and can hint me in a direction to further analyze or rectify this?

alexmuller, 2023-11-22:
I am also having this problem with an Inky 7.3

I have a very small `main.py`:

```
import time

from picographics import PicoGraphics, DISPLAY_INKY_FRAME_7 as DISPLAY

# A short delay to give USB chance to initialise
time.sleep(0.5)

# Setup for the display.
graphics = PicoGraphics(DISPLAY)
WIDTH, HEIGHT = graphics.get_bounds()
graphics.set_font("sans")

graphics.set_pen(1)
graphics.clear()
graphics.set_pen(0)

graphics.set_thickness(2)

graphics.text('Hello', 20, 40, 600, 2)
graphics.text("Today this thing is happening", 20, 90, 600, 1)

graphics.text('Shopping list', 20, 160, 600, 2)
graphics.text('Thing one', 20, 210, 600, 1)
graphics.text('Thing two', 20, 240, 600, 1)

print('Updating graphics')
graphics.update()
print('Done')
```

In the Thonny output I see:

```
MPY: soft reboot
Updating graphics
```

If I disconnect the frame from USB and reconnect it later on, I can never get `graphics.update()` to execute successfully.

If I reflash the Micropython with the uf2 download it's fine again, until I disconnect it from USB.

w3stbam, 2024-01-29:
Good news! Seems i have found and fixed the issue. As we identified some kind of blocking behaviour i had a deeper dive into the code. I modified the busy_wait function to return after a given timeout:

https://github.com/pimoroni/pimoroni-pico/compare/main...w3stbam:pimoroni-pico:fixupdate

This way, a block will be removed after a timeout of currently 30 seconds and the code continues. I am not really sure why the block has not been removed before, but anyways, this seems to fix thei issue. I have now ~ 20 refreshes on battery with a 100% success rate.
If you want to try the fix, you can download the firmware here:
https://github.com/w3stbam/pimoroni-pico/releases/download/1.21.1/firmware.uf2

If it works for you, give some feedback here then i can make a pull request against the main repo

klausi83, 2024-01-30:
Thank you w3stbam. Great work. I have installed the firmware you linked and am currently undergoing tests. I will send you a my report tomorrow, after I have seen it working for a while.

klausi83, 2024-01-31:
Great Work w3stbam. Your fix seems to have done the trick. During my first trials I had 2 freeze-ups which might be totally unrelated to this thread. After that my inky-frame has been running flawlessly on battery for 12h straight with an update every 20 minutes and occasional button-usages on my side to give me extra picture updates.
Perfekt!

w3stbam, 2024-01-31:
Sounds great @klausi83 . Regarding the freezes you have experienced, these may be related to the timeout having a too low value. I changed it from 30 to 45 seconds after a comment on the pimoroni forums, where a user said that refreshes may take up to 40 seconds. You can try the version with 45 seconds timeout here, it should be somewhat safer:
https://github.com/w3stbam/pimoroni-pico/releases/download/1.21.2/pimoroni-inky_frame-1.21.2-micropython.uf2