pimoroni/pimoroni-pico #870: pfguitil seems to get the font sizes wrong

ay4, 2023-11-03:
As I've been trying out the great [pfguitil](https://github.com/gadgetoid/pgfutil) to create my custom fonts, I invariably failed. Looking at the console, it seems that pfguitil gets the char sizes wrong.

**Example 1** 

[image]

As I'm loading this example, which is a 128x72 px PNG file, I'm getting the following console output:

```
(112, 56) 6 6
Chessboard 7x7
Loading image 128x72, Char size 7x8
Chessboard 8x9
``` 

and the following picture:

[image]

...which then duly translates into the wrong .bitmap font.

**Example 2**

[image]

This is a 8x16 font embedded into a 128x128 PNG file. Loading it results in the following console output:

```
(112, 56) 6 6
Chessboard 7x7
Loading image 128x128, Char size 7x15
Chessboard 8x16
122 78
```

and the following outcome:

[image]

In this case, it's less evident that something is broken, but the bottom line of block-drawing characters is shifted one pixel down (as are other lines, but it's unnoticable).

**What I tried:**

- Setting up chessboard size before loading PNG (the tool ignores set size)
- Changing PNGs to be indexed palette (no influence on the outcome)
- Using transparent background (all symbols become black rectangles)

So, is there a workaround maybe? Probably not as it seems to be off-by-one error around the usage of CHAR_WIDTH / CHAR_HEIGHT in core.py. However I'm not a good enough engineer to suggest a fix.

Gadgetoid, 2024-03-11:
I think the util probably needs fixing, or some TLC in general. I haven't come back to it since I threw it together as an experiment, and it's definitely clearly both very useful and very broken!

Gadgetoid, 2024-03-11:
Had a crack at this, and also fixed another bug relating to byte packing sizes, see: https://github.com/Gadgetoid/pgfutil/pull/3

Discussion should resume in that PR or at https://github.com/Gadgetoid/pgfutil/issues/2