pimoroni/pimoroni-pico #871: i75 + Pico W ever increasing memory usage

Ben-Logibase, 2023-11-06:
Hi there, running an i75 + Pico W combo. Developed a script only to have it crash randomly after a few hours every time. Upon debugging, found an ever increasing RAM usage when updating the display. Thought it was my code but now I see that it could be a problem with your underlying i75 lib. Running your example at: https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/interstate75/raw_set_pixel.py produces the same problem. If you "import gc" and then add print(f"Memory: used:{gc.mem_alloc()} b free:{gc.mem_free()} b") below your print("Setting Pix... you will see the runaway RAM usage - even after the .clear() call. Also, manually running gc collection doesn't seem to help either. Could you please advise.

Gadgetoid, 2023-12-12:
This is odd, I can't see anything in the i75 code that would allocate memory and cause a leak. It all boils down to very trivial set-a-value-in-a-buffer operations.

All `clear` does is set all the pixels in the buffer to 0, so it wouldn't release any RAM.

If you run `raw_set_pixel.py` and remove `h75.set_pixel(x, y, r, g, b)`, does it still leak?

Gadgetoid, 2025-04-25:
Closing for now. Please re-open over at https://github.com/pimoroni/interstate75 if you're still having trouble.

closed 2025-04-25 by Gadgetoid