pimoroni/pimoroni-pico #872: ServoCluster for Servo2040 board hangs execution

moomurrs, 2023-11-07:
I'm using the Servo2040 Micropython v1.21.0. The Servo class works perfectly and I'm getting ready to use the ServoCluster but instancing a ServoCluster hangs the entire program.

I'm following the example [here ](https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/servo#getting-started-1) on the REPL interpreter as follows:
`>>> from servo import ServoCluster, servo2040`
`>>> pins = list(range(servo2040.SERVO_1, servo2040.SERVO_4 + 1))`
`>>> cluster = ServoCluster(0, 0, pins) # <-- hangs here`

But the last line ServoCluter hangs execution in the interpreter. The same behavior is observed if put in main.py. 
I understand the ServoCluster is using PIO state-machine but I'm not using the anything else that's using the state-machine. I've tried switching switching the pio=1 or sm=1 such as 
`cluster = ServoCluster(1, 0, pins)`
`cluster = ServoCluster(0, 1, pins)`
 but there's no difference.
I've tried downgrading the firmware to pico-v20.6 but no difference.
This hanging is observed on both Thonny and Mu IDE on Windows 11.

This is my exact output on Thonny:
`>>> from servo import ServoCluster, servo2040`
`>>> pins = list(range(servo2040.SERVO_1, servo2040.SERVO_4 + 1))`
`>>> cluster = ServoCluster(0, 0, pins)`
`| # <-- cursor hangs here`

I can see micropython successfully imports the ServoCluster class:
`>>> from servo import ServoCluster`
`>>> ServoCluster`
`<class 'servo_cluster'>`

If I start typing when it hangs, Thonny complains about a timeout:
`PROBLEM IN THONNY'S BACK-END: Exception while handling 'execute_source' (serial.serialutil.SerialTimeoutException: Write timeout).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.`

I'm hoping I'm missing something simple but I still cannot seem to create a ServoCluster object in Micropython. Everything else works just (LEDs, which uses the state machine and Servo Class are working). Currently, 5V line is cut, so the board is powered externally by 5V battery.

Any ideas? Is this reproducible by anyone else?
Thank you.

ZodiusInfuser, 2023-11-07:
Hi @moomurrs,

This is an odd one. I have just grabbed a Servo 2040 from my shelf, flashed it with the v1.21.0 build for Pico, and then ran your REPL lines, but it did not hang. I even put the Pico W build on (as that does use PIO), and it went through fine.

```python
MicroPython v1.21.0, picow v1.21.0 on 2023-10-06; Raspberry Pi Pico W with RP2040

Type "help()" for more information.

>>> from servo import ServoCluster, servo2040
>>> pins = list(range(servo2040.SERVO_1, servo2040.SERVO_4 + 1))
>>> cluster = ServoCluster(0, 0, pins)
>>> print(cluster)
ServoCluster(servos = {
	{pin = 0, enabled = False, pulse = 0.0, value = 0.0, phase = 0.0}, 
	{pin = 1, enabled = False, pulse = 0.0, value = 0.0, phase = 0.25}, 
	{pin = 2, enabled = False, pulse = 0.0, value = 0.0, phase = 0.5}, 
	{pin = 3, enabled = False, pulse = 0.0, value = 0.0, phase = 0.75}
}, freq = 50.0)
>>> 
```

Can I just double check which exact build of our Micropython you are using (i.e. the full file name)? Also, when you ran your REPL lines, was that from a fresh power-on or had you run some code before that? It could be that something had not been cleaned up by Micropython, so this is worth a try if you have not done so already.

When did you purchase your Servo 2040? I'm not discounting a hardware issue at this point, though it seems unlikely to me.

moomurrs, 2023-11-07:
Hi!
Thank you for getting back so quickly to assist me.

The full name is:
`MicroPython v1.21.0, pico v1.21.0 on 2023-10-06; Raspberry Pi Pico with RP2040`, used this firmware file:
`pimoroni-pico-v1.21.0-micropython.uf2` for the Servo 2040
The hanging is observed regardless of hard reset (power disconnect) or soft reset (IDE 'Stop'), i.e. no other code was running before. I've even tried gc.collect() from this [example ](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/servo2040/servo_cluster.py). I've also tried: 
`pins = [1]` 
to be more clear but no dice.

I currently have all 18 servos hooked up. They all respond perfectly with Servo class but servo 2 & 18, and servo 1 & 17 are tied together. I'm so close to the finish line 😭

Only way to get the board to respond is to press **RESET** on the board or power-cycle. IDE soft-reboot 'Stop' doesn't work when it hangs; I get this error:
```Unable to connect to COM5: could not open port 'COM5': PermissionError(13, 'Access is denied.', None, 5)
If you have serial connection to the device from another program, then disconnect it there first.
Process ended with exit code 1.
```

I have a 5v regulator that only outputs 5v (no pot to increase voltage). The board never smoked and shows no signs of damage.
I'm also using a power blocker (data lines intact, but power line cut) on the USB. I've tried both USB 2.0 and USB 3.0 ports without hubs. No dice. 
I just tried disconnecting the battery and connecting the board directly to USB 3 (no power blocker). No change either, still hanging.

I did another firmware wipe for 1.21.0 , just in case but still hanging.

moomurrs, 2023-11-07:
Hi,

I've made some progress!
I flashed an incorrect firmware (pico**w** v 1.21.0) and the ServoCluster did **not** hang! But I noticed it wiped out all my py files (no worries, backed up on git). 
I've now flashed the correct firmware (pico v 1.21.0) and the ServoCluster is still working just fine. 
This makes me think whatever py file I had in storage was causing conflicts which is weird because I do NOT have a file or class called ServoCluster or servo2040. I'll slowly re-add my project files until I find the culprit. 

But for now, the issue is resolved.!

Thank you so much for pointing me in the right direction to flash the picoW firmware (this will wipe your py files but fixed the issue). I will mark resolved.

closed 2023-11-07 by moomurrs