pimoroni/pimoroni-pico #874: Hub75 drivers question

crashlab, 2023-11-08:
Hi,
I've bought the interstate75 w board and I've started to play with it.
I've looked at the source code of the [Hub75 driver](https://github.com/pimoroni/pimoroni-pico/blob/main/drivers/hub75/hub75.cpp) and I wasn't able to figure out why in the hub75.cpp - inside Hub75::start (at line 150):

```
// Same handler for both DMA channels
irq_set_exclusive_handler(DMA_IRQ_0, handler);
irq_set_exclusive_handler(DMA_IRQ_1, handler);

dma_channel_set_irq0_enabled(dma_channel, true);

irq_set_enabled(pio_get_dreq(pio, sm_data, true), true);
irq_set_enabled(DMA_IRQ_0, true);
```

the function "irq_set_exclusive_handler" is called for DMA_IRQ_1 but only for DMA_IRQ_0 the function "irq_set_enabled" is called.

However, it gets then disabled inside Hub75::stop (at line 169):

```
irq_set_enabled(DMA_IRQ_0, false);
irq_set_enabled(DMA_IRQ_1, false);
irq_set_enabled(pio_get_dreq(pio, sm_data, true), false);
```

Is it really needed to set an exclusive handler for DMA_IRQ_1?

Gadgetoid, 2023-11-08:
Might be a vestigial holdout from an attempt at chained DMA, but either way it looks superfluous.

crashlab, 2023-11-09:
@Gadgetoid Thanks a lot for confirming that. I'll create a pull request soon to remove the unnecessary code.

closed 2023-11-09 by crashlab