pimoroni/pimoroni-pico #875: Single pin encoder

Chick92, 2023-11-09:
Hello, 

Fist of all thankyiu for your incredible contributions to the open source community. I've made extensive use of your encoder and motor / motor2040 code. Unfortunately the encoders / motors I'm using are very poor quality and are causing a lot of erratic behavior.

I'm intended to switch the motors / encoders to a unit with a built in driver which provides a single encoder output. I.e the the "motor" has 5 pins, power, ground, pwm input, direction input and encoder output.

My question is if the encoder library supports this? 

Thanks!

Ben

ZodiusInfuser, 2023-11-09:
Hi Ben,

Glad to hear you have found our motor / motor2040 code of value.

I am not familiar with encoders that would output just a single PWM signal. It must be an absolute angle, as you would not be able to determine the direction of travel otherwise. Do you have a datasheet you can link so I can confirm?

In any case, our Encoder library does not support this type of encoder, as it is specifically for those that output a quadrature signal. The RP2040 seems to support reading of single PWM signals, but this is not exposed nicely in Micropython. The best I have found with 10 seconds of googling is this repo: https://github.com/phoreglad/pico-MP-modules/blob/main/PWMCounter/PWMCounter.py

As a side note, you mention the motor is controlled by a PWM and a direction input. If you have not already seen, our Motor class supports this mode (but is not the default). https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/motor#driver-type

Hope that helps

Chick92, 2023-11-09:
@ZodiusInfuser Hey, sorry I don't think I was clear in what i meant. The encoder feedback is just a single channel, it's not PWM. This is an example of a similar motor (and one I'm using for development) -

https://cdn.robotshop.com/rbm/a00a7635-653b-4220-aac9-b0c23c5c5e2c/d/d86e25ee-9cba-4fd3-93ee-8538436c9ad4/3f672a79_25SG-2418BL-12V.pdf

https://www.robotshop.com/products/24mm-brushless-dc-gear-motor-12v-100rpm

ZodiusInfuser, 2023-11-09:
@Chick92 Thanks for those links.

PWM was my guess of what the output could have been given it was on a single wire (though it could have been UART or analog too). Looking at that datasheet I was not far off, as that encoder outputs a PFM signal, which is a pulse signal with the frequency proportional to the speed of the motor (if I am inferring the meaning of FG correctly)

Our library does not support this kind of output, but that PWMCounter.py I linked to before may be able to decode it.

Also, our enviro grow product has moisture sensors that produce a similar output, which we decode using this function:
https://github.com/pimoroni/enviro/blob/b5bff82f9001409c9683be43b9f2114e7da8b430/enviro/boards/grow.py#L27-L64

Hope that helps.

closed 2023-11-09 by ZodiusInfuser

Chick92, 2023-11-09:
That's really helpful, thanks very much!