pimoroni/pimoroni-pico #887: pico_graphics is not memory safe

jondurrant, 2023-12-22:
frame_buffer is allocated with new but never destroyed even if the pico_graphics object is removed.

Gadgetoid, 2024-01-16:
Good spot- MicroPython handles `frame_buffer` allocation so we've not had much hitting this code path. Looks like, at least, it needs a `managed` flag and a corresponding `delete` in the class destructor.

There's also a method to replace the framebuffer, I'm not totally sure if that should be blocked if `managed == true` or just let the user get themselves into trouble- 

https://github.com/pimoroni/pimoroni-pico/blob/911cbb710ebb725d83065a21438c35783ef8b19c/libraries/pico_graphics/pico_graphics.cpp#L26-L28

jondurrant, 2024-01-17:
Yes I moved to create my own buffers to make it memory safe. 

I'd really recommend having a full unit test suite for these libraries. Perhaps use CPPUTest which will do the memory safe analysis for you.  I have videos and examples of how on YouTube (https://youtu.be/_2-Mh3Kun6s) which might be an accelerator.

Gadgetoid, 2024-01-17:
I'd love unit testing across the libraries but I fear it might be an effort of quickly diminishing returns for a project of this scope- it's always been on my to-look-in-to-list though.

closed 2024-06-04 by Gadgetoid