pimoroni/pimoroni-pico #889: I2C pins are undocumented in motor2040 header file

robberwick, 2024-01-01:
The `motor2040.hpp` file declares variables for all the connected pins on the board, but omits definitions for the pins connected to the SDA/SCL pins of the QW/ST connector (GPIO20 & GPIO21 respectively). These values can be found by looking at the schematics, but It would be really useful to add these so that it's possible to access them via the `motor::MOTOR2040` namespace.

ZodiusInfuser, 2024-01-08:
Hi @robberwick,

Thank you for raising this. The reason these pins were not defined in the `motor::MOTOR2040` namespace is because they are accessible through the `I2C` class within `pimoroni_i2c.hpp`. This is our convenience class for dealing with I2C comms, which simplifies some of the operations involved in reading and writing registers.

One of its constructors accepts `BOARD` enum value from `pimoroni_common.hpp` of which MOTOR_2040 is one of those already defined. Does this work for you? If not I will have a look at your PR.

robberwick, 2024-01-08:
Hi @ZodiusInfuser 

Ok, if that's the convention, then I guess it's worth sticking to that. I would just say that as a user of that board, I wasn't aware of that mechanism, so I would suggest that it probably needs to be more discoverable. For my money, I would still define the pins in that header file, since that seems to be the most obvious place to go looking for them (at least it seemed that way to me), but perhaps also add a comment on there to point the user at the i2c helper + board definition documentation so that they can choose which is most appropriate for them.

If that's how you decide to go, and you want me to try to write a suitable comment, I can update my PR accordingly.

ZodiusInfuser, 2024-02-22:
Closing this as the associated PR is now merged.

closed 2024-02-22 by ZodiusInfuser