pimoroni/pimoroni-pico #891: Can't find climits when trying to compile a program with pico_explorer

bda65hp, 2024-01-06:
Hi all,
I can't compile anything with pico_explorer.
I can't find climits somewhere (not in pico-sdk nor pico-extras ...)

`In file included from /home/bruno/src/pico/pimoroni-pico/drivers/motor/motor.hpp:5,
                 from /home/bruno/src/pico/pimoroni-pico/libraries/pico_explorer/pico_explorer.hpp:3,
                 from /home/bruno/src/pico/scd30_lib/scd30_lib.c:40:
/home/bruno/src/pico/pimoroni-pico/common/pimoroni_common.hpp:3:10: fatal error: climits: No such file or directory
    3 | #include <climits>
      |          ^~~~~~~~~
compilation terminated.`
Is there something weird on my installation? FYI i'm using alpinelinux edge.
I'm using en pico-sdk and pico-extras since some weeks with great success. I've buy a pico-display and want to add it to my own program (with a sensiron scd30 sensor which work well when reporting values on usb tty).

Thanks

Gadgetoid, 2024-01-12:
What compiler are you using?

This might be related - https://forums.pimoroni.com/t/pico-c-trying-to-use-the-pico-display-2/19231

bda65hp, 2024-01-12:
Hi,
I've removed alpinelinux and returned to fedora today.
Compilation work now with success (but i need to rewrite my code in C++ as there are many C++ class in pimoroni libs.
Sorry for polluting here...
Thanks

closed 2024-01-12 by bda65hp