pimoroni/pimoroni-pico #894: Micropython jpegdec::get_width()/get_height() will always return 0

Skyler84, 2024-01-12:
Due to the [`// Just-in-time open of the filename/buffer`](https://github.com/pimoroni/pimoroni-pico/blob/bff245324b6375c4d97c84ec7421d1dc4189d1eb/micropython/modules/jpegdec/jpegdec.cpp#L247), get_width and get_height will always return 0 as the jpeg is never actually loaded when these functions are called. This means the user cannot check the size of the jpeg and change the x/y/scale when decoding it based on the size.

This should probably be fixed by loading the jpeg at the time of calling `open_file()` or `open_RAM()`, which would also want a `close()` function.

Gadgetoid, 2024-01-12:
You're right- PNGDEC rectifies this (and avoids the need for close) by opening the file to grab the width/height and store it on the Python object instance- 

https://github.com/pimoroni/pimoroni-pico/blob/bff245324b6375c4d97c84ec7421d1dc4189d1eb/micropython/modules/pngdec/pngdec.cpp#L292-L321

This pattern was never backported to JPEGDEC, but probably should be.

Skyler84, 2024-01-12:
I've changed #895 to backport the PNGDEC fix, adding width/height to the _JPEG_obj_t and setting those on open(), creating a helper function `jpegdec_open_helper(self)` and using that in the open functions and decode. I've also reverted my previous changes so it's just the backport (the MP close() binding is removed etc)