pimoroni/pimoroni-pico #896: pico crashes writing VL53L5CX data to USB

rokroskar, 2024-01-16:
Hi there, apologies if this is not the right place for such questions - happy to take it elsewhere if that is the case. 

I am trying to stream data from the VL53L5CX to a computer attached with USB with this code: 

Pico (using the pimoroni demo code as a starting point): 

```
import micropython
import sys
import ustruct

import breakout_vl53l5cx
import pimoroni_i2c

PINS_BREAKOUT_GARDEN = {"sda": 4, "scl": 5}
i2c = pimoroni_i2c.PimoroniI2C(**PINS_BREAKOUT_GARDEN, baudrate=2_000_000)

sensor = breakout_vl53l5cx.VL53L5CX(i2c)
sensor.set_resolution(breakout_vl53l5cx.RESOLUTION_4X4)
sensor.set_ranging_frequency_hz(1)
sensor.start_ranging()

fmt = 'i'*16
rlen = ustruct.calcsize(fmt)
buf = bytearray(rlen)

while True:
    if sensor.data_ready():
        data = sensor.get_data()
        ustruct.pack_into(fmt,buf, 0, *data.distance)
        sys.stdout.buffer.write(buf)
```

Mac:
```
import serial
import struct

fmt = 'i'*16
rlen = struct.calcsize(fmt)

s = serial.Serial(port="/dev/tty.usbmodem11101", baudrate=115200, parity=serial.PARITY_EVEN, stopbits=serial.STOPBITS_ONE)
s.flush()

while True:
    res = s.read(rlen)
    print(struct.unpack_from(fmt,res))
```

A few things:

1. I've tested the serialization / comms with another code that just sent random integers - this worked fine with no errors
2. When I run the pico part through the REPL, I eventually get `Exception occured (maybe a connection loss)`
3. When I upload the pico code as `main.py` and run it without the REPL, the Pico never seems to connect (I can't see the usb device, nor does it allow the IDE to connect). The only way to get back into the Pico is to disconnect the sensor (presumably so the code crashes sooner). 

I'm new to using the Pico so I'm at a loss as to how to debug this further or what might be causing the crashes. Note that in the case of a "crash" the pico actually disconnects, so it appears to be some device issue?

Gadgetoid, 2024-01-16:
I wonder if this is something relating to the Pico not sleeping long enough for the USB interrupts to be serviced.

Might be worth trying -
1. Add a `time.sleep(0.1)` in your `while True` loop
2. Try the (very experimental) MicroPython build from #892 (:warning: will trash your Pico's filesystem if you try to revert to an earlier build, so back up files first)

This said, I don't know why random integers would work where this does not- in theory the same problem should apply.

rokroskar, 2024-01-16:
Thanks @Gadgetoid for the tip, that might very well be it. My other code had a `time.sleep(1)` in there... in my head I was thinking it will block on `sensor.data_ready()` but of course it does not.

rokroskar, 2024-01-16:
I think that fixed it, thanks @Gadgetoid! The pico does take an unusually long time to come up with this simple script, but it does work in the end.

closed 2024-01-16 by rokroskar