pimoroni/pimoroni-pico #902: Virtual PicoGraphics frontend

martinfiers, 2024-02-04:
Hello,

I'm not sure if this is the best place to put this question - feel free to move to a better place.

When experimenting / building UIs with PicoGraphics, one limitation is the refresh rate of ~ 40s for some e-ink displays (I'm using the inky frame, which is awesome btw!).

For rapid prototyping, it would be very useful if one could us esome sort of virtual display (on your PC) with an identical API is the PicoGraphics - i.e., with the ability to write text (same fonts), draw lines, circles, etc. On the PC the results would be instant, hence giving a nice write/test/view results loop. At the same time you don't need to connect your hardware, which is great for development / prototyping purposes. Once your code works, you can upload it to your pico hardware and run it - hopefully giving the same results :)

Is something like this being considered?

Gadgetoid, 2024-02-27:
I've toyed with this idea a few times, and set it up for something else (PicoSystem) out of curiosity. It involved pairing SDL with the UNIX port of MicroPython and figuring out how to reliably do all of the pixel format conversion.

Wow... that was 3 years ago now - https://github.com/pimoroni/picosystem/commits/micropython-sdl/

Getting the API of the UNIX port of MicroPython to match our much-more-complicated Pimoroni Pico set of libraries (PicoSystem was very concise) will probably be the trickier part, since it would be nice for examples written for one of our boards to Just Work on the desktop version.

So it's sort-of being considered, and it's definitely possible, but it's not something I have the time to do at the moment. It's quite involved!

heavyimage, 2024-10-01:
Just found this page with the same question @martinfiers had.  Would be very cool to prototype code on a PC and then deploy (with a few tweaks) to some picographics hardware!