pimoroni/pimoroni-pico #908: picographics - fixed_width parameter missing from method text()

swm11, 2024-03-10:
The readme for picographics provides the following signature:
```
display.text(text, x, y, wordwrap, scale, angle, spacing)
```

The text after that also describes the fixed_width parameter:
```
fixed_width - space all characters equal distance apart (monospace)
```

However, as far as I can determine, this hasn't been implemented but it would be useful.  I'm testing on a Badger 2040 W  (wonderful device!!!).

Gadgetoid, 2024-03-11:
I think the `fixed_width` argument is implemented, if you supply the keyword argument `fixed_width=True` to `display.text()` it should work.

There were some fixes to it for Badger - https://github.com/pimoroni/badger2040/issues/51

swm11, 2024-03-11:
Thanks for the feedback.  My mistake - I was using an out-of-date distribution on my Badger-W.

In the PicoGraphics README.md you might like to change:
```
display.text(text, x, y, wordwrap, scale, angle, spacing)
```
to
```
display.text(text, x, y, wordwrap, scale, angle, spacing, fixed_width)
```

closed 2024-03-11 by swm11