pimoroni/pimoroni-pico #913: emojis support

RamblingGeekUK, 2024-03-23:
Could a font be added to support emojis?

Gadgetoid, 2024-03-25:
This is something that PicoVector (a vector graphics library for Pico) would be best at and, in theory, yes it could. Though they would be basic - FontAwesome-like - single colour icons. Actually I think the font tools could convert FontAwsome itself... 🤔 

PicoVector is due for a significant rewrite, though, before it's ready for prime time.

RamblingGeekUK, 2024-03-25:
Great, would that work today?   I'm working on the idea of using a 2 x 64 x 32 HUB75 Matrix with a InterState75 to display messages from Twitch Chat.. so would be good if it could display emotes.

Gadgetoid, 2024-03-25:
> Great, would that work today?

Probably not... 

To even get started you'd have to wrangle the font converter from this old version of Alright Fonts to convert some emoji codepoints from a suitable font. I haven't touched this stuff in a while so I don't have any insightful pointers - https://github.com/lowfatcode/alright-fonts/tree/e79124b71416fda85fd5e6439f87520b7f3cec8f

You'd probably be better off using PNG images if you want it done fast. Messing with fonts is going to be a *journey*!