pimoroni/pimoroni-pico #914: Brightness support for 5x5 RGB Matrix Breakout

alexraileanu, 2024-03-25:
Hello,

I'm trying to use the Micropython library for the [5x5 RGB Matrix Breakout](https://shop.pimoroni.com/products/5x5-rgb-matrix-breakout?variant=21375941279827) but I can't seem to find a way to set the brightness for an individual led. I saw in the [examples](https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/breakout_rgbmatrix5x5/demo.py) that there's also no mention of this. Am I missing something?

Thanks!

Gadgetoid, 2024-03-25:
The Python library just scales the individual colour channels by the brightness value, so you could do something like this:

```
brightness = 0.5
matrix.set_pixel(x, y, int(r * brightness), int(g * brightness), int(b * brightness))
```

To achieve the same result.

Brightness control is indeed currently unsupported in the Pico libraries.

alexraileanu, 2024-03-25:
That seems to work, thanks a lot @Gadgetoid !

I could try to make a PR to add brightness control (at least for the 5x5 matrix) but I'm not super sure where or how to start. Do you perhaps have any pointers for me?

Thanks again!

Gadgetoid, 2024-03-25:
Most of our Python modules are implemented in C++ so it's frustratingly complex. The main thrust of the RGB 5x5 set_pixel is here:

https://github.com/pimoroni/pimoroni-pico/blob/6eb0f90e53e503790150028f2cf181961f589619/libraries/breakout_rgbmatrix5x5/breakout_rgbmatrix5x5.cpp#L27-L36

Which calls:

https://github.com/pimoroni/pimoroni-pico/blob/6eb0f90e53e503790150028f2cf181961f589619/drivers/is31fl3731/is31fl3731.cpp#L83-L85

So ideally here it would multiply the R, G and B channels by a brightness value from 0-255, scale it back to the range 0-255 and then apply gamma using the `GAMMA_8BIT` table from "common/pimoroni_common.hpp"

That brightness value would then have to be passed through from the Python bindings:

https://github.com/pimoroni/pimoroni-pico/blob/6eb0f90e53e503790150028f2cf181961f589619/micropython/modules/breakout_rgbmatrix5x5/breakout_rgbmatrix5x5.cpp#L49-L85

You can add brightness as an optional keyword arg with something like:

```
{ MP_QSTR_brightness, MP_ARG_INT, {.u_int = 255I} },
```

If you fork `pimoroni-pico` on GitHub, you can push changes up and - with any luck- it'll build you a new MicroPython without you needing to grapple with a local build environment.

Using C bindings for libraries like this was a choice we made to avoid having to write everything twice and it's definitely had its pitfalls. Your time might be better spent trying to get the pure Python library to work 😬

alexraileanu, 2024-03-25:
Thanks for the very detailed reply, @Gadgetoid ! I'm not a CPP kinda guy but I'll try!

> Your time might be better spent trying to get the pure Python library to work

Trust me I spent far longer on this project than I should have and now that I've found a way to make it work I'm not sure I want to debug that further :p. 

Some struggles I had:

- Started with TinyGo on an Arduino Nano IoT 33. There was no lib for this 5x5 matrix so I copied the one you guys made into Go (which was a gigantic pain but I got there in the end)
- For another device I have, I also couldn't find a library in TinyGo and tried to write my own, failed quite badly. 
- Changed to an Arduino ESP 32 because of the Micropython support hoping that I would be able to get these Python libraries to work. Failed miserably.
- Now finally bought an RPi Nano W, flashed it with the Pimoroni Pico Micropython and finally I have both devices working nicely

Anyway, sorry for the rambling! I'll give the brightness thing a shot somewhere this week. Thanks again!

Gadgetoid, 2024-03-25:
>  flashed it with the Pimoroni Pico Micropython and finally I have both devices working nicely

As contrived and troublesome as our libraries-included MicroPython build is, it's nice when it just has all the things good to go. And good to hear it was your solution!

No time like the present to dive into C++ 😆

closed 2025-01-09 by thirdr