pimoroni/pimoroni-pico #941: Add jpg decode options

exussum12, 2024-05-21:
https://github.com/pimoroni/pimoroni-pico/blob/f3b53b6b5ffd37202fc0de4fab4f159d551b7a24/libraries/jpegdec/JPEGDEC.h#L220C9-L220C15

seems to be an older version

https://github.com/bitbank2/JPEGDEC/pull/55/files

Added support for offsetting

Gadgetoid, 2024-05-21:
OOh good catch, this is a useful feature for dealing with single file sprite, icon or tile sheets. Though jpeg is still a little suboptimal for those in general.

Gadgetoid, 2024-06-04:
Uhhh.. The linked PR was rejected and those changes aren't actually part of JPEGDEC at all...

I wish I'd noticed this before I wasted my time :skull:

closed 2024-06-04 by Gadgetoid

exussum12, 2024-06-04:
Ah sorry :( I was looking to add upstream and the PR seemed good. I didn't spot it was rejected either :(

Gadgetoid, 2024-06-04:
S'okay, I've got an open sync branch for when we get some crop functionality :laughing: 

Didn't realise until I'd updated the bindings, tried to compile, and thought "wut!? Y Error!?"

exussum12, 2025-05-12:
[image]

The upstream support has 100% landed this time and are documented in the wiki

https://github.com/bitbank2/JPEGDEC/wiki