pimoroni/pimoroni-pico #987: MicroPython Libray for Pico Display 1.14"

FabLabUtoPi, 2024-08-13:
Hi
in 2021 i published an article about Pico Display 1.14" on my blog :https://www.framboise314.fr/utiliser-les-extensions-pimoroni-avec-raspberry-pi-pico/#Pico_Display_un_ecran_LCD_16021580
for this product https://shop.pimoroni.com/products/pico-display-pack?variant=32368664215635
At this time there was a library in your github : https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/pico_display
Actually this link gives a 404 error
i tried to find this lib. but cannot find it 
can you help ? 
thank you
francois / www.framboise314.fr
contact@framboise314.fr

Gadgetoid, 2024-08-13:
All of our display products were merged into PicoGraphics, see: https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/picographics

Here's an example for Pico Display: https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_display/rainbow.py

And using buttons changed to `Button()` so it's common across all products, eg:

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_display/button_test.py

FabLabUtoPi, 2024-08-13:
Thank you Philip
 i look at this and update my article
have a nice day
francois

closed 2024-10-02 by helgibbons