pimoroni/pimoroni-pico #988: Request for additional resolutions  DISPLAY_INTERSTATE75_192X32  DISPLAY_INTERSTATE75_256X32

menges404, 2024-08-17:
Hello!
great work! Is there a chance to get the resolution 192x32 (3x 64x32) i made some fine test for home automation and telephone display with 64x32 and 128x32 an build some alu-frames for 3x 64x32 panels bevor i realise, that this resolution is current unavailable - is this implementabe?

menges404, 2024-08-19:
Sorry, i re-read my post - ans here some additional explanations: "Is there a chance" and "is this implementabe" in the meaning of: "yes, some additional defines" (i'll try) and "recompile" (i actually do not know) and "tests are nessasary" (i can do)
or "no, due to memory, or other limitation we must reduce the count of resolutions" 
helps for my next steps.

Gadgetoid, 2024-08-19:
IIRC memory/flash overhead of additional resolutions isn't too bad.

I would really like it all to be dynamic, so you can just use whatever resolution and pixel layout you desire- but that's actually quite tricky with the PIO driver!

Definitely do try tinkering yourself, since if you fork this repo GitHub will spit out a build. I am :mountain_snow: snowed under :mountain_snow: ... as usual!

Gadgetoid, 2025-04-25:
Linked from https://github.com/pimoroni/interstate75/issues/32 to track this, but closing here since the Interstate 75 builds have moved.

We have now have two 128x64 panels arranged in a 128x128 layout (i75 Duo) which shows the extra fiddling required to make 64x32x2 into a 64x64 layout. In short we really, really do need some kind of arbitrary "map this graphics region to the panel" setup.

closed 2025-04-25 by Gadgetoid

alphanumeric007, 2025-04-25:
I managed to get 64x32x2 into a 64x64 layout. Hel Gibbons pointed me to the interstate75.py file.
To which I added
```
        if display == DISPLAY_INTERSTATE75_64X64:
            out_width = 128
            out_height = 32 
```
A variation of the custom 128x128 already in there.
```
        if display == DISPLAY_INTERSTATE75_128X128:
            out_width = 256
            out_height = 64
```
And then saved the edited file to my i75 RP2350.

Gadgetoid, 2025-04-27:
It never occurred to me that the hack for that would work for another resolution but it was written *fairly* resolution agnostic. That’s good news and we should probably throw together an example 🤣 I’ve been tunnel vision on Duo lately because it’s good fun!

alphanumeric007, 2025-04-30:
I got lucky in that my panels are half that size both ways. 64x32 versus 128x64. And it was a simple edit. =) 

I spied the Duo75 references and poked around a bit trying to figure out what it was. It's fun some times sleuthing out what something is from the files in the repository. It looks like its a Double Mint Gum deal, double your pleasure double your fun, lol. Two PR2350's one per panel connector. Bigger Better **Faster**