pimoroni/pimoroni-pico #990: Inky frame rotate

mrglennjones, 2024-08-19:
When using rotate=90 on the inky frame 7.3, it actually rotates it 180

Gadgetoid, 2024-08-19:
🤔 https://github.com/pimoroni/pimoroni-pico/blob/970046e84ae73cad23deae28fd6e96427aa0eb64/drivers/inky73/inky73.cpp#L88

Looks like it doesn’t support 90/270 degree rotations (some mix of the chip not supporting it and/or it being a pain to accomplish in software) so if the rotation != 0 then it’s set to 180.

Might be worth looking into why 90 isn’t supported- of course past me didn’t leave any useful comment 😫

ticky, 2025-06-02:
just ran across this, +1 for supporting 90-degree rotation, given the rest of picographics doesn’t readily support drawing rotated!