pimoroni/pimoroni-pico #996: Pico display isn't working :( / I'm a bad programmer

I-cannot-code, 2024-09-12:
I've recently purchased and recieved my pico, and I've believe I set it up right; put it in bootloader mode, downloaded the uf2 file and downloaded thonny. However, every time I try to run code on it to show something on the screen, nothing works! I tried around 1 week ago and it worked, however I forgot what I did and now nothing appears on the screen, it just says 'invalid syntax', but I've copied the raw code from Github!
Some advice would be helpful, thank you!
[image]

Gadgetoid, 2024-09-12:
It looks like you're trying to run the C++ examples (that's on us, they're much easier to find in this repository than the MicroPython ones) in Thonny, which needs Python code.

Try some of the .py examples from https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_display

helgibbons, 2024-10-02:
Closing this one for now - let us know if you're still stuck!

closed 2024-10-02 by helgibbons