pimoroni/pimoroni-pico #997: pico-v1.23.0-1-pimoroni-micropython flashing on Pico 2

knlklabacka, 2024-09-12:
I'm not sure what I'm doing wrong and hoping you can help? I have downloaded pico-v1.23.0-1-pimoroni-micropython.uf2 from your repo, to my Windows 10 machine, copied the file to the RP2350 device under file explorer.  The file looks like it copies but does not automatically reboot  the Pico 2 once the copy is complete.  I have tried unplugging and plugging the Pico 2 back in with no luck of having Pico 2 flashed.  The Pico 2 just shows up in Windows as RP2350 device.  If you need more information please let me know.

Gadgetoid, 2024-09-12:
(ﾉ◕ヮ◕)ﾉ*:・ﾟ✧  https://github.com/pimoroni/pimoroni-pico-rp2350/releases/tag/v0.0.7

(yes we should probably add a link to these in the README!)

knlklabacka, 2024-09-12:
Thank you @Gadgetoid.  It is working now!

closed 2024-09-12 by knlklabacka