pimoroni/pimoroni-pico #999: Font Creation

peardox, 2024-09-19:
A load of hte samples use [fonts](https://github.com/pimoroni/pimoroni-pico/tree/main/libraries/bitmap_fonts)

Are these generated by some common app ? If so what is it? Specifically some app that outputs the CPP + HPP ?

If it's common then it should be mantioned in an easily accessible place

If not (and there is no app that does close to the cpp+hpp) then I guess I could write one - but don't wanna re-invent the wheel

Hoping there's something already (I'm lazy)

Gadgetoid, 2024-09-19:
There’s PGFUtil but it’s intended to create fonts for loading in MicroPython, though it shouldn’t be a huge stretch to convert those to hpp. It’s also in desperate need of TLC - https://github.com/gadgetoid/pgfutil

peardox, 2024-09-19:
Yeah, that was an extrapalation of the desired functionality. Language awareness - would be easy to make one that uses a template for max flexibility...

peardox, 2024-09-30:
Found some Chinese utility but - well, it's all Chinese

So - started rolling my own. Finally got as far as being able to render a TTF/OTF accurately to the screen and then on to a bitmap

[image]
This is a sample 32pt Segoe UI font (Win System font) - note the max rect is 33 high (from 32pt? fonts...)

[image]
This one's dogica - a 8pt render of a free font for small systems (need to DL + zoom image I guess)

The grey checkerbox is the rectangle for the char and the pink one is where the actal TTF gets rendered to
Could have an option to force monospace so e.g. Segoe was all same width
Space could be saved by only storing the pink area + an offset (possibly not worth it?)
Could also add in colour option for things like Emotes and hand-drawn glyphs

Currently Windows, Mac + Linux is Easy (FMX allows all mainstream OSes). Pi a little harder but do-able (I think) - would need Lazarus

Slion, 2024-12-03:
I've been using [BMFont](https://www.angelcode.com/products/bmfont/) in the past to generate textures for OpenGL. Not sure how applicable it will be to Pimoroni bitmap fonts.

Slion, 2024-12-03:
Found a tool to convert bmfont to cop;
https://github.com/bibendovsky/bmfont_to_cpp