pimoroni/pimoroni-pico #1000: Example pico_display scroll_menu_h.py not working

yene, 2024-09-21:
hi,

the examples
micropython/examples/pico_display/scroll_menu_h.py
micropython/examples/pico_display/scroll_menu_v.py

do not work at all on my pico display. 
The text does not fit into the display and the button presses are not recognised.

yene, 2024-09-21:
from https://github.com/pimoroni/pimoroni-pico/pull/982 
@thirdr

anomaly256, 2024-09-25:
Hi, just stumbled across the same issue.  I managed to fix the buttons by importing Button from pimoroni, changing the setup of the buttons to use this library, and changing the `button_x.value == 0:` tests to `button_x.read():`.

For example, for the 'simple_menu' demo:

```python
import time
from pimoroni import Button
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2, PEN_RGB565
from pimoroni import RGBLED
from machine import Pin

button_a = Button(12)
button_b = Button(13)
button_x = Button(14)
button_y = Button(15)

display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, pen_type=PEN_RGB565, rotate=0)
display.set_backlight(0.8)

# set up constants for drawing
WIDTH, HEIGHT = display.get_bounds()
BLACK = display.create_pen(0, 0, 0)
RED = display.create_pen(255, 0, 0)
GREEN = display.create_pen(0, 255, 0)
BLUE = display.create_pen(0, 0, 255)
WHITE = display.create_pen(255, 255, 255)
PURPLE = display.create_pen(255, 0, 255)

led = RGBLED(26, 27, 28)

class Menu(object):
    def __init__(self):
        self.items = ["Red", "Green", "Blue", "Purple"]
        self.selected = 0
        self.shadow_offset = 2
        self.cursor = "<-"
        self.colour = BLACK
        self.title = "Simple Menu"

    # A function to draw only the menu elements.
    def draw_menu(self):
        display.set_pen(WHITE)
        display.clear()
        display.set_pen(BLACK)
        length = display.measure_text(self.title, 4)
        display.text(self.title, WIDTH // 2 - length // 2 + self.shadow_offset, 10 + self.shadow_offset, WIDTH, 4)
        display.set_pen(self.colour)
        display.text(self.title, WIDTH // 2 - length // 2, 10, WIDTH, 4)

        display.set_pen(BLACK)
        for item in range(len(self.items)):
            length = display.measure_text(self.items[item], 3)
            if self.selected == item:
                display.set_pen(self.colour)
                display.text(self.cursor, length + 40, HEIGHT // 2 + item * 20, WIDTH, 3)

            display.text(self.items[item], 30, HEIGHT // 2 + item * 20, WIDTH, 3)

            display.set_pen(self.colour)
            display.rectangle(0, HEIGHT - 10, WIDTH, 10)
            display.set_pen(BLACK)

    # Do a thing based on the currently selected menu item
    # For our example we'll be changing the text and LED colour
    def process_selected(self):
        if self.selected == 0:
            self.colour = RED
            led.set_rgb(255, 0, 0)

        if self.selected == 1:
            self.colour = GREEN
            led.set_rgb(0, 255, 0)

        if self.selected == 2:
            self.colour = BLUE
            led.set_rgb(0, 0, 255)

        if self.selected == 3:
            self.colour = PURPLE
            led.set_rgb(255, 0, 255)

    def user_input(self):
        # Process the user input and update the currently selected item
        if button_y.read():
            print("Y")
            if self.selected + 1 < len(self.items):
                self.selected += 1
            else:
                self.selected = 0

        if button_x.read():
            print("X")
            if self.selected > 0:
                self.selected -= 1
            else:
                self.selected = len(self.items) - 1

        if button_a.read():
            print("A")
            self.process_selected()

        if button_b.read():
            print("B")

menu = Menu()

while True:

    menu.draw_menu()
    menu.user_input()

    display.update()

    time.sleep(1.0 / 20)
```

yene, 2024-09-26:
@anomaly256 thank you very much, I can work with this.

thirdr, 2024-09-27:
Hello! :)

I've resolved the issue with the buttons not working and raised a PR, which can be found here: #1003

Regarding the other issue of it incorrect displaying, this example is designed specifically for use with the Pico Display 2 and 2.8, and it won't display properly on the smaller Pico Display. I should have made that clearer in my code, so I've now added a comment to the top of both examples.

closed 2024-10-02 by helgibbons