pimoroni/pimoroni-pico #1001: pico_display_demo sleep (why?)

peardox, 2024-09-23:
I just made a small tweak to the pico_display demo for the pico display pack

I noticed the code has sleep_ms(1000 / 60);

I set a variable to time_us_64() at the start of the main loop then calculated the elapsed time (i.e. frame time) at the end of the loop where the sleep is

The elapsed time was in the 62700 us region (fluctuates slightly) which is about 16 FPS

The sleep adds an extra 16667 us to the frame and is far less than the implied 60FPS it's seemingly trying to limit

OK, some screens could potentially go > 60FPS but surely it'd be better to at least check if the frame time was under the limit before adding more delay? Better in such cases would be to calc a more accurate value for the sleep....

Gadgetoid, 2024-09-26:
It looks like a relatively old example into which not much thought was put. Could probably use some TLC, albeit I'd reserve that for after the PicoVector rewrite has been merged: https://github.com/pimoroni/pimoroni-pico/pull/853

Accurate frame pacing is probably somewhat outside of the scope of a simple API demo, though, but could be nice in a standalone example.

peardox, 2024-09-26:
Removing the code that draws a load of trangles to create a sort of windmill effect so you've just got the bouncing balls left and removing the sleep increases the speed to > 60FPS - making it a bit more ralistic

But, yeah, the display demos prob need a bit of an overhaul

peardox, 2024-10-03:
Covered more fully in other issues

closed 2024-10-03 by peardox