pimoroni/pimoroni-pico #1002: Cosmic Unicorn: White noise when playing audio and accessing the REPL

bmustill-rose, 2024-09-24:
Tested with "MicroPython v1.23.0, cosmic_unicorn v1.23.0-1 on 2024-06-06; Raspberry Pi Pico W" - unfortunately I don't have any other appropriate boards to test with.

Steps:

1. Flash this https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/cosmic_unicorn/audio/simple_playback.py example and supporting files.
2. Reboot the device.
3. Audio plays as expected.
4. Access the REPL.
5. Control + c if required.
6. Control + d to reboot the device.
7. White noise is played and the board freezes / REPL becomes unresponsive.

Gadgetoid, 2024-09-24:
This is somewhat a known-issue for which I have made an upstream fix to MicroPython, but it wont take effect in downstream code until the next MicroPython point release: https://github.com/micropython/micropython/pull/15585

It's somewhat possible to workaround this by using a `try / catch / finally` block around the i2s audio to explicitly call `deinit` but it shouldn't be necessary.

Gadgetoid, 2024-09-24:
In the short term I think this should work:

```python
from audio import WavPlayer

sound = WavPlayer(0, 10, 11, 9, amp_enable=22)

sound.play_wav("beepboop.wav", False)

try:
    while sound.is_playing():
        pass

finally:
    sound.__stop_i2s()
```

Will treat this issue as a bug, however, and leave it open until this is *actually* fixed.

bmustill-rose, 2024-09-24:
That appears to be working well and will do for now as you say. Thanks.