pimoroni/pimoroni-pico #1005: Pico2 slows down display pack

peardox, 2024-10-01:
Yeah, I couldn't believe it either...

Soldered up 2x Pico 2 and 1x Pimoroni Pico Plus 2 to try them out on diplsy packs

On my Pico Ws the display_pack_2_demo runs at 30fps (I added frame timing)

On all Pico 2 devices the FPS drops tro 24fps - tried all three devices I have to hand

Tested on a display pack 2.8 - I have the others as well but doubt there'll be much change (apart from the smallest as it has a smaller screen), the 2 + 2.8 are just a bigger LCD, exact same pixel counts

peardox, 2024-10-01:
Ahh - just spotted #985

@Gadgetoid I'll stick my code on Github in a little while and add a note here

peardox, 2024-10-01:
Here you go - https://github.com/peardox/display_pack_28

Gadgetoid, 2024-10-01:
I'm happy if you want to make a pull-request of it and claim the glory :laughing:

I had assumed you were using MicroPython, so I was waaaaaay off.

peardox, 2024-10-01:
I was already thinking of doing a PR with the modifications just for the display_pack_28 definitions but also adding the FPS demo will be nicer

I wanna check out the SPI rate as (150/133) * 62,500,000 |= 75,000,000 (actual value is a funny number so wanna experiment first)

peardox, 2024-10-02:
And yet changing it to...

```
//PicoGraphics_PenRGB332 graphics(st7789.width, st7789.height, nullptr);
PicoGraphics_PenRGB565 graphics(st7789.width, st7789.height, nullptr);

```

Dropped the frame rate from 45.x to 43.5-ish

peardox, 2024-10-02:
Oops 888 crashes - mem requirements are 240k for that res but I am on a Pico2 so plenty to spare

Looks like a different bug...

Slion, 2024-10-02:
> Dropped the frame rate from 45.x to 43.5-ish

That's weird, maybe there is something else at play. You better look at the frame time or even just the update time when trying to optimise your display performance. Though FPS is still interesting.

peardox, 2024-10-02:
Tried Pen4, Pen8, Pen332, and Pen565 which all work

Only Pen888 dies horribly (Pen4 just looks horrible)

Slion, 2024-10-02:
> Pens other than 565 all run thru a frame conversion function

Yup

peardox, 2024-10-02:
Now things get even weirder - switched to a PicoW and 332 is 30spf while 565 is 33fps (i.e. faster as you imagined it would)

The opposite rate change the pico2 

I also added a method to return the actual SPI rate. The Pico1 won't go above 62,5Mhz and the Pico2 75Mhz

I'll have to switch back to a Pico2 to see what waas reported from a 62.5Mhz request - which according to #985 should be 37.5