pimoroni/pimoroni-pico #1007: Full repo build doesn't work on rp2350

peardox, 2024-10-01:
The removal of the RTC in Pico2 is causing some demos to fail on these devices

Attempting a full repo build with Ninja cases this error in breakout_encoder_wheel

> /home/simon/pimoroni-pico/examples/breakout_encoder_wheel/clock/clock.cpp:6:10: fatal error: hardware/rtc.h: No such file or directory
>     6 | #include "hardware/rtc.h"
>       |          ^~~~~~~~~~~~~~~~
> compilation terminated.
> [778/15238] Building CXX object examples/breakout_encoder_...se_game.dir/__/__/__/drivers/ioexpander/ioexpander.cpp.obj
> ninja: build stopped: subcommand failed.

Gadgetoid, 2024-10-01:
There’s a fix for this - and other issues with SDK 2.0.0 - in one of the open PRs here- just requires an include change. I’ve been quite focussed in on the MicroPython stuff so it hasn’t made its way into main yet.

AIUI there’s another SDK revision coming this month… maybe… possibly… so I’m not in a tremendous hurry 😬

peardox, 2024-10-01:
OK, the Ninja build broke when I was testing my PR for pico_display_28 + SPI so I just built mine with Make to test and wante to point it out in case it was unreported

Gadgetoid, 2024-10-02:
Collating those fixes here - https://github.com/pimoroni/pimoroni-pico/pull/1008